serialport/node-serialport #2127: parser-inter-byte-timeout does not included in the @serialport

huangxuewu, 2020-08-10:
## 📚 Documentation

(A clear and concise description of how the docs could be better, with links if possible)

### Have you read the [Contributing Guidelines on docs](https://github.com/serialport/node-serialport/blob/master/CONTRIBUTING.md#writing-documentation)?

(Write your answer here.)

when installing lastest version via yarn, the parser-inter-byte-timeout file does not include

reconbot, 2020-08-11:
This is correct, you can install it yourself but I'd also merge a PR adding it.

huangxuewu, 2020-08-14:
yeah, I thought you maybe forget to add it. since you already noticed I will close this issue.

closed 2020-08-14 by huangxuewu

reconbot, 2020-08-14:
Want to make a pr?

---
Francis Gulotta
wizard@roborooter.com

PR #2202 (2021-01-20): Add missing inter-byte-timeout parser to package
Add missing inter-byte-timeout parser to package

this will resolve #2127