serialport/node-serialport #2129: Problem with prebuild ia32

s00d, 2020-08-17:
### Summary of Problem

Where did all the prebuild packages for ia32 for [9.0.1](https://github.com/serialport/node-serialport/releases/tag/%40serialport%2Fbindings%409.0.1)? 

```

### Versions, Operating System and Hardware

- SerialPort@? 9.0.1
- Node.js v? 14.7.0
- Windows? Linux? Mac? Mac
- Hardware and chipset? (Prolific/FTDI/Other) FTDI

reconbot, 2020-08-17:
32bit support for linux and osx has been dropped by NodeJS, and we're having a lot of trouble getting it to run for windows.

s00d, 2020-08-18:
Will you no longer support ia32 for Windows? Can you add at least 9.0.1 for Windows? Please

reconbot, 2020-08-18:
I have tried for many hours but have not been successful.

s00d, 2020-08-19:
on lts, ​​everything should work. Maybe just downgrade to version 12 or is there a reason to upgrade to 14?

reconbot, 2020-08-20:
The issues are with Appveryor, it's 32 bit windows machines time out without logging any errors.

s00d, 2020-08-21:
problem connected with node 14 and x32.
if you change node to lts appveyor wil bi work good.
[Example](https://github.com/s00d/node-serialport/blob/master/appveyor.yml) 
`nodejs_version: "12"`
you can add --verbose to npm install
node 14 not normally work in x32
[image]

you can add build a package for Linux

I created a [request](https://github.com/serialport/node-serialport/pull/2132), please add

juliocesaroliveira, 2020-09-17:
I used node@13.8.0 to build Windows ia32 resources.
The generated resources worked fine.

s00d, 2020-09-21:
problem only with version 14, can be switched to 13 version. @reconbot can you fix it?

reconbot, 2020-10-16:
There wasn't a problem with 14, the problem was with appveyor

closed 2020-10-20 by s00d