serialport/node-serialport #2131: Errors/Warnings while creating a Docker image with Alpine Linux

mn567, 2020-08-20:
Getting Errors/Warning in a Docker project using Alpine Linux

(Please answer all 3)

Creating a Docker image using Alpine Linux
Getting compilation warnings
Should build without warnings

This is the Docker file #1 which gives warnings (Docker file #2 is at the end)

FROM node:12.18.3-alpine3.12 as builder

COPY package*.json ./

RUN apk add --no-cache make gcc g++ python2 linux-headers udev

RUN npm install serialport --unsafe-perm --build-from-source=serialport

RUN npm ci --only=production

FROM node:12.18.3-alpine3.12

WORKDIR /opt/adapters/Modbus

COPY --from=builder node_modules .

COPY . .

# Add Tini
ENV TINI_VERSION v0.19.0
ADD https://github.com/krallin/tini/releases/download/${TINI_VERSION}/tini-static /tini
RUN chmod +x /tini
ENTRYPOINT ["/tini", "--"]

CMD [ "node", "index.js" ]
```

The following are the warnings

 docker build -t modbus:1.0 .
Sending build context to Docker daemon  18.31MB
Step 1/14 : FROM node:12.18.3-alpine3.12 as builder
 ---> 73e8a4cf7fa8
Step 2/14 : COPY package*.json ./
 ---> Using cache
 ---> fc00e97c16e4
Step 3/14 : RUN apk add --no-cache make gcc g++ python2 linux-headers udev
 ---> Running in 90cc1d086c3e
fetch http://dl-cdn.alpinelinux.org/alpine/v3.12/main/x86_64/APKINDEX.tar.gz
fetch http://dl-cdn.alpinelinux.org/alpine/v3.12/community/x86_64/APKINDEX.tar.gz
(1/30) Upgrading musl (1.1.24-r8 -> 1.1.24-r9)
(2/30) Installing binutils (2.34-r1)
(3/30) Installing gmp (6.2.0-r0)
(4/30) Installing isl (0.18-r0)
(5/30) Installing libgomp (9.3.0-r2)
(6/30) Installing libatomic (9.3.0-r2)
(7/30) Installing libgphobos (9.3.0-r2)
(8/30) Installing mpfr4 (4.0.2-r4)
(9/30) Installing mpc1 (1.1.0-r1)
(10/30) Installing gcc (9.3.0-r2)
(11/30) Installing musl-dev (1.1.24-r9)
(12/30) Installing libc-dev (0.7.2-r3)
(13/30) Installing g++ (9.3.0-r2)
(14/30) Installing linux-headers (5.4.5-r1)
(15/30) Installing make (4.3-r0)
(16/30) Installing libbz2 (1.0.8-r1)
(17/30) Installing expat (2.2.9-r1)
(18/30) Installing libffi (3.3-r2)
(19/30) Installing gdbm (1.13-r1)
(20/30) Installing ncurses-terminfo-base (6.2_p20200523-r0)
[... 590 lines cut ...]
FROM node:12.18.3-alpine3.11 as builder

COPY package*.json ./

RUN apk add --no-cache make gcc g++ python linux-headers udev

RUN npm install serialport --unsafe-perm --build-from-source=serialport

RUN npm ci --only=production

FROM node:12.18.3-alpine3.12

WORKDIR /opt/adapters/Modbus

COPY --from=builder node_modules .

COPY . .

# Add Tini
ENV TINI_VERSION v0.19.0
ADD https://github.com/krallin/tini/releases/download/${TINI_VERSION}/tini-static /tini
RUN chmod +x /tini
ENTRYPOINT ["/tini", "--"]

CMD [ "node", "index.js" ]

scyto, 2021-02-26:
how did you fix this?

MBerka, 2021-06-21:
@scyto Someone would have to go through the serialport C++ and resolve all the compiler warnings to avoid this noise, but an inverse `grep` on the build command could filter them out too.