serialport/node-serialport #2135: Error in detecting port path automatically.

miteshdube, 2020-08-25:
I am new to node-serialport and having some issues in detecting serial port automatically. Below is my code,

```
var fs = require('fs');
var serialport = require('serialport');
var usb = require('usb');
var portName = " ";
```

Code for detect serial port path automatically:-
(Note: If I tried to print portName on console is prints exact port path e.g. COM4,COM3,etc. But getting error in opening port. See error at the end. And if I assign portName as hard coded, like `var portName = "COM4";` then it runs successfully).

```
serialport.list().then(
  ports => {
    ports.forEach(port => {
    if(port.manufacturer.includes("Silicon Labs")&&port.pnpId.includes("USB\\VID_10C4&PID_EA60\\0001")){
    console.log("Port Set");
    portName = port.path;
    }
   })
  },
  err => {
   console.error('Error listing ports', err)
  }
 );
``` 

Code to read data from serial port:-

```
var myPort = new serialport(portName, {baudRate:9600});
var Readline = serialport.parsers.Readline;
var parser = new Readline();
myPort.pipe(parser);
myPort.on('open', showPortOpen);
parser.on('data', readSerialData);
myPort.on('close', showPortClose);
myPort.on('error', showError);
function showPortOpen() {
  console.log('port open. Data rate: ' + myPort.baudRate);
}
function readSerialData(data) {
  console.log(data);
  return;
}

function showPortClose() {
  console.log('port closed.');
}

function showError(error) {
  console.log('Serial port error: ' + error);
}
```
Error in opening:

`Serial port error: Error: Opening : Unknown error code 123`

jsftsylvain, 2020-08-25:
Hey Miteshdube, 
I am having a similar problem. Seems like you aren't opening the right path though, log your path before you open it to check it. Keep in mind that since SerialPort.list() is an async function the code where you open your path might execute before it updates the port path.

Look at my stack overflow post below for a different [issue](https://stackoverflow.com/questions/63530909/declaring-a-new-serialport-as-a-variable-doesnt-allow-me-to-read-data-serialpo) related to yours.

miteshdube, 2020-08-26:
Thanks for your response. I solved my issue please see on my stack overflow [post](https://stackoverflow.com/questions/63572949/error-in-detecting-serial-port-path-automatically).

closed 2020-09-11 by reconbot