serialport/node-serialport #2137: Unintended unicode "replacement character" within the data stream

ecrode, 2020-09-06:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?
I'm using a [serial port interface (link)](https://github.com/node-red/node-red-nodes/tree/master/io/serialport) within node.red which relies on this library.
- What happens?
When ever an invalid character is received, it is replaced with [EF BF BD] sequence (original byte is lost). Searching I determined that it represents a Unicode ["Replacement Character" (link)](https://apps.timwhitlock.info/unicode/inspect?s=%EF%BF%BD). This issue occurs for characters with MSB (most significant bit) set as long as they are among normal (ascii) characters. Binary data which is not contained among the regular string does not appear to exhibit the issue.
Issue appears similar to unsolved [issue #1209 (link)](https://github.com/serialport/node-serialport/issues/1209)
- What should have happened?
Original byte / data are left as is

### Code to Reproduce the Issue

```js
const serialport = new SerialPort(path)

// Code
```

### Versions, Operating System and Hardware

- SerialPort@?
- Node.js v 12.18.3
- OS: Linux / raspberrypi
- Hardware and chipset: FTDI

ecrode, 2020-09-06:
I'll have to think about code to reproduce the issue, but you can try creating a loop-back socket and sending "Hello World" while having any of the characters in the middle having a MSB set.

ecrode, 2020-09-06:
Here is an example of data corruption:
https://jsonblob.com/b8d60da6-f005-11ea-ae37-d9193694cf04

The payload is the string after node red processing, extra spaces are removed as well as MSB of the individual characters cleared, that is why the corruption shows up as "o?=" whereas a single ":" with MSB set was transmitted. The controller in question uses MSB as an indicator that this character should be blinking.

Note that raw data is included, the string takes place from position [4] to position [46], corruption occurs from 33 to 35 whereas three characters are transmitted whereas only one ":" & 0x80 = 3A & 0x80 = 0xBA = 186  was transmitted.

reconbot, 2020-09-06:
I know it's difficult with node red but example core would help a lot. Generally the replacement character happens with line noise. This feels like something set the encoding of the stream to ascii instead of buffers.

ecrode, 2020-09-06:
I apologize, found the issue within my code which was dealing with buffer as string, thereby adding an error. I should really skip late night coding sessions 🙇

closed 2020-09-06 by ecrode

reconbot, 2020-09-06:
Happens to us all

DykiSA, 2022-06-04:
For me the buffer is forced to the string with predefined encoding, very annoying

adma05, 2024-04-21:
Help pls, how to fix this?

[image]