serialport/node-serialport #2139: i think install serial port with electron but i found error "compiled against a different Node.js version"

maxpromer, 2020-09-09:
Hello, I need to install serialport with electron, here is `package.json`

```json
  "devDependencies": {
    "electron": "^9.2.0",
    "electron-rebuild": "^2.0.2"
  },
  "dependencies": {
    "bcrypt": "^5.0.0",
    "serialport": "^9.0.1"
  }
```

after i run `npm install` i found

```
> @serialport/bindings@9.0.1 install D:\GitHub\microBlock-IDE-offline\node_modules\@serialport\bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

> bcrypt@5.0.0 install D:\GitHub\microBlock-IDE-offline\node_modules\bcrypt
> node-pre-gyp install --fallback-to-build

node-pre-gyp WARN Using request for node-pre-gyp https download
[bcrypt] Success: "D:\GitHub\microBlock-IDE-offline\node_modules\bcrypt\lib\binding\napi-v3\bcrypt_lib.node" is installed via remote

> core-js@3.6.5 postinstall D:\GitHub\microBlock-IDE-offline\node_modules\core-js
> node -e "try{require('./postinstall')}catch(e){}"

Thank you for using core-js ( https://github.com/zloirock/core-js ) for polyfilling JavaScript standard library!

The project needs your help! Please consider supporting of core-js on Open Collective or Patreon:
> https://opencollective.com/core-js
> https://www.patreon.com/zloirock

Also, the author of core-js ( https://github.com/zloirock ) is looking for a good job -)

> electron@9.2.0 postinstall D:\GitHub\microBlock-IDE-offline\node_modules\electron
> node install.js

> serialport@9.0.1 postinstall D:\GitHub\microBlock-IDE-offline\node_modules\serialport
> node thank-you.js

Thank you for using serialport!
If you rely on this package, please consider supporting our open collective:
> https://opencollective.com/serialport/donate

> microblock-ide-offline@1.0.0 install D:\GitHub\microBlock-IDE-offline
> electron-rebuild

√ Rebuild Complete
added 293 packages from 293 contributors and audited 293 packages in 17.247s
found 0 vulnerabilities
```

but after i run npm start i found

```
> microblock-ide-offline@1.0.0 start D:\GitHub\microBlock-IDE-offline
> electron .

App threw an error during load
Error: The module '\\?\D:\GitHub\microBlock-IDE-offline\node_modules\@serialport\bindings\build\Release\bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 72. This version of Node.js requires
NODE_MODULE_VERSION 80. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.func [as dlopen] (electron/js2c/asar.js:140:31)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1034:18)
    at Object.func [as .node] (electron/js2c/asar.js:140:31)
    at Module.load (internal/modules/cjs/loader.js:815:32)
    at Module._load (internal/modules/cjs/loader.js:727:14)
    at Function.Module._load (electron/js2c/asar.js:769:28)
    at Module.require (internal/modules/cjs/loader.js:852:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at bindings (D:\GitHub\microBlock-IDE-offline\node_modules\bindings\bindings.js:112:48)
    at Object.<anonymous> (D:\GitHub\microBlock-IDE-offline\node_modules\@serialport\bindings\lib\win32.js:1:36)
Terminate batch job (Y/N)? ํ
Terminate batch job (Y/N)? Y
```

so i don't know why i found error and How to fixed it ?

Thanks.

maxpromer, 2020-09-09:
OK, for fixed this i download `bindings-v9.0.1-electron-v80-win32-x64.tar.gz` then replace `bindings.node` at `\node_modules\@serialport\bindings\build\Release`

jacobq, 2020-09-11:
Sounds like you were able to resolve the problem. Mind closing the issue? For most cases [`electron-rebuild`](https://github.com/electron/electron-rebuild) should also do the trick.

pardovot, 2020-09-13:
> Sounds like you were able to resolve the problem. Mind closing the issue? For most cases [`electron-rebuild`](https://github.com/electron/electron-rebuild) should also do the trick.

I also encountered this today and electron-rebuild fixed the issue for me.

maxpromer, 2020-09-14:
> Sounds like you were able to resolve the problem. Mind closing the issue? For most cases [`electron-rebuild`](https://github.com/electron/electron-rebuild) should also do the trick.

if you see post me log, you will find `electron-rebuild` run by `npm install`

jacobq, 2020-09-14:
> > Sounds like you were able to resolve the problem. Mind closing the issue? For most cases [`electron-rebuild`](https://github.com/electron/electron-rebuild) should also do the trick.
> 
> if you see post me log, you will found `electron-rebuild` run by `npm install`

Indeed, I did. However, sometimes it is not able to detect the electron version correctly plus may incorrectly determine that no rebuild is required. [`electron-rebuild --version x.y.z --force`](https://github.com/electron/electron-rebuild#cli-arguments) Bottom line: if the problem shows a mismatch in the `NODE_MODULE_VERSION` then the solution almost certainly involves recompiling the module.

pardovot, 2020-09-14:
> Indeed, I did. However, sometimes it is not able to detect the electron version correctly plus may incorrectly determine that no rebuild is required. [`electron-rebuild --version x.y.z [...]

As you said for me npm install didn't solve the issue while electron-rebuild did.

maxpromer, 2020-09-16:
> Indeed, I did. However, sometimes it is not able to detect the electron version correctly plus may incorrectly determine that no rebuild is required. [`electron-rebuild --version x.y.z [...]

Thank you, i will try next times. and Thank you for work hard.

closed 2020-09-16 by maxpromer