serialport/node-serialport #2140: Support Electron 10

panuhorsmalahti, 2020-09-16:
## 💥 Proposal

### What feature you'd like to see

Support for Electron 10.x.

## Motivation

node-serialport should support latest version of Electron.

This can likely be accomplished simply by rebuilding and releasing this package.

JuliadsonS, 2020-09-23:
Alguma novidade? funciona perfeitamente com a versão 8.2.0 do eléctron!

ukasiu, 2020-11-25:
this will never become stale :)

mkurczewski, 2020-11-27:
still nothing?

RHeynsZa, 2020-12-02:
Any workaround? Other than downgrading electron

orange-beans, 2020-12-05:
Electron 10 and Electron 11 not working.

reconbot, 2020-12-05:
I released a serialport with the latest abi support a few days ago.

orange-beans, 2020-12-06:
> I released a serialport with the latest abi support a few days ago.

Thank you for the reply. The latest binding I can find for v9.0.3 is v80 (bindings-v9.0.3-electron-v80-win32-x64.tar.gz);
However, the error message I encountered said it requires NODE_VERSION_V85.

My configuration is as following:
Electron: v11.0.3
Node:v12.16.1
serialport: v9.0.3

Anything I'm missing?

Thanks with regards.

reconbot, 2020-12-17:
Lets try it again with 9.0.4

reconbot, 2020-12-17:
Looks like we're having build issues on linux and windows. I'm digging into it.

reconbot, 2020-12-17:
I've ported the builds to github actions, seems to be in good shape.

orange-beans, 2020-12-18:
> I've ported the builds to github actions, seems to be in good shape.

Thanks for the speedy update! Now the Electron App can successfully be loaded with _serialport_ binding.

However, a new problem has been encountered as shown below. (which has never been seen in version 8.x.x)

The configuration:
Electron: v11.0.3
Node:v12.16.1
serialport: v9.0.3 (Binding version V85)

I tried to create a new port with following code:
```Javascript
const testPort = new SerialPort('COM1', defaultPortConfig, function(err) {
  if (err) console.log(err);
  else {
    console.log('Sucess')
  }
});
```

And the following error pops out:
```Javascript
TypeError: Third argument must be a function
    at internal/util.js:297:30
    at new Promise (<anonymous>)
    at open (internal/util.js:296:12)
    at WindowsBinding.open (D:\Projects\Desktop-projects\Electron-Boilerplates\electron-boilplate-v140\app\node_modules\@serialport\bindings\lib\win32.js:56:22)
```

It is strange that in the code third argument is indeed a function. I tried with arrow function, the same error still pops up.

Lovinity, 2020-12-19:
I am getting the same error as @orange-beans

Lovinity, 2020-12-19:
I'm having the same issue as @orange-beans 

Opened a new issue at https://github.com/serialport/node-serialport/issues/2191

GazHank, 2021-07-05:
New Electron versions are now supported and published

closed 2021-07-05 by GazHank