serialport/node-serialport #2141: SerialPort.list() only shows ports when script is run with sudo (Rasperry Pi)

mikelyndersOKCC, 2020-09-18:
# Summary of Problem
- What are you trying to do?

I have developed an Electron Application that displays/graphs/analyzes data from several lab instruments with serial outputs. I have an "autoconnect" feature that use SerialPort.list() to get a list of the ports which I then try to query for serial numbers. This works fine on Windows (which I was using to develop) but now I am trying to deploy my application on several Raspberry Pi's.

- What happens?

SerialPort.list() returns an empty list `[]` when run in both Electron and in a minimal test script (see below) on the Raspberry Pi. However, I noticed that running the same script with `sudo` works as expected. Interestingly, I can connect to the port directly without `sudo` if I know the port name ahead of time ex: */dev/ttyUSB0*  

Also, `$ udevadm info -q path -n /dev/ttyUSB0` returns nothing but returns as expected with sudo : `$ sudo udevadm info -q path -n /dev/ttyUSB0`

Any ideas what's going on here? I imagine I shouldn't need `sudo` to get a list of ports, especially if I can connect to them without it. Also, as a sanity check, I ran a python script with pyserial which lists the ports without `sudo` on the same Raspberry Pi, so I don't think I have a weird udev/permissions setup.

- What should have happened?

SerialPort.list() should return a list of ports even when the script is run without `sudo`

### Code to Reproduce the Issue

```js
const SerialPort = require('serialport')

SerialPort.list().then(ports=>console.log(ports), err=>console.log(err))

```

### Versions, Operating System and Hardware

- SerialPort@9.0.1
- Node.js v14.11.0
- Linux (Raspberry Pi 4) Raspberry Pi OS August 2020
- Hardware and chipset? RPi -> USB/RS-232 Adapters -> Generic Serial Device (Laboratory Instrument)

fivdi, 2020-09-19:
The code shown above will list the available ports if run by user `pi` on Raspberry Pi OS. If a different user runs the script, that user will need to be a member of group `gpio` in order for the code function correctly. If the program is being run by a user that is not a member of group `gpio` please add the user to group `gpio` to see if it works.

fivdi, 2020-09-19:
Note that to actually use the serialport, the user will also need to be a member of group `dialout`.

mikelyndersOKCC, 2020-09-21:
Hello, I can confirm that I ran the script as user `pi` which is in both `gpio` and `dialout` and it returned an empty list. Only SerialPort.list() behaves differently between `pi` and `sudo`. All the other functions work correctly under user `pi` including connecting, reading, and writing serial ports (provided you know the name of the port ahead of time).

Currently I am using a workaround that `ls \dev | grep USB` to search the \dev directory for devices (all my usb-serial adapters get named ttyUSBXX).

fivdi, 2020-09-21:
Below is what I see on Raspberry Pi OS (32-bit) Lite August 2020 when one serial cable is connected to a Pi 4 via USB. In other words, I can't reproduce the error.
```
pi@raspberrypi:~/serialport $ whoami
pi

pi@raspberrypi:~/serialport $ groups
pi adm dialout cdrom sudo audio video plugdev games users input netdev gpio i2c spi

pi@raspberrypi:~/serialport $ uname -a
Linux raspberrypi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux

pi@raspberrypi:~/serialport $ cat /etc/debian_version 
10.4

pi@raspberrypi:~/serialport $ npm list | grep serialport
/home/pi/serialport
└─┬ serialport@9.0.1
  ├─┬ @serialport/binding-mock@9.0.1
  │ ├─┬ @serialport/binding-abstract@9.0.1
  ├─┬ @serialport/bindings@9.0.1
  │ ├── @serialport/binding-abstract@9.0.1 deduped
  │ ├── @serialport/parser-readline@9.0.1 deduped
  ├── @serialport/parser-byte-length@9.0.1
  ├── @serialport/parser-cctalk@9.0.1
  ├── @serialport/parser-delimiter@9.0.1
  ├─┬ @serialport/parser-readline@9.0.1
  │ └── @serialport/parser-delimiter@9.0.1 deduped
  ├── @serialport/parser-ready@9.0.1
  ├── @serialport/parser-regex@9.0.1
  ├─┬ @serialport/stream@9.0.1
npm ERR! extraneous: readable-stream@3.6.0 /home/pi/serialport/node_modules/bl/node_modules/readable-stream
npm ERR! extraneous: readable-stream@3.6.0 /home/pi/serialport/node_modules/tar-stream/node_modules/readable-stream

pi@raspberrypi:~/serialport $ udevadm info -q path -n /dev/ttyUSB0
/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.0/ttyUSB0/tty/ttyUSB0

pi@raspberrypi:~/serialport $ node -v
v14.9.0

pi@raspberrypi:~/serialport $ cat t.js 
const SerialPort = require('serialport')

SerialPort.list().then(ports=>console.log(ports), err=>console.log(err))

pi@raspberrypi:~/serialport $ node t.js 
[
  {
    manufacturer: 'FTDI',
    serialNumber: 'XXXXXXXX',
    pnpId: 'usb-FTDI_FT232R_USB_UART_XXXXXXXX-if00-port0',
    locationId: undefined,
    vendorId: '0403',
    productId: '6001',
    path: '/dev/ttyUSB0'
  },
  {
    manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    locationId: undefined,
    vendorId: undefined,
    productId: undefined,
    path: '/dev/ttyAMA0'
  }
]
```
I guess there is something different about your installation but I'm not sure what.

- Is the new 64-bit versions of Raspberry Pi OS being used?
- How was the image on the SD card created? Was an old version updated or was Raspberry Pi OS August 2020 freshly flashed onto the SD card?
- Were permissions modified?
- Are new udev rules needed for the serial cable being used?

mikelyndersOKCC, 2020-09-21:
I really appreciate you looking into this @fivdi 
Here's the output on my end:
```console
pi@raspberrypi:~/Desktop/serialTest $ whoami
pi
pi@raspberrypi:~/Desktop/serialTest $ groups
pi adm dialout cdrom sudo audio video plugdev games users input netdev gpio i2c spi
pi@raspberrypi:~/Desktop/serialTest $ uname -a
Linux raspberrypi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux
pi@raspberrypi:~/Desktop/serialTest $ cat /etc/debian_version
10.4
pi@raspberrypi:~/Desktop/serialTest $ npm list | grep serialport
└─┬ serialport@9.0.1
  ├─┬ @serialport/binding-mock@9.0.1
  │ ├─┬ @serialport/binding-abstract@9.0.1
  ├─┬ @serialport/bindings@9.0.1
  │ ├── @serialport/binding-abstract@9.0.1 deduped
  │ ├── @serialport/parser-readline@9.0.1 deduped
  ├── @serialport/parser-byte-length@9.0.1
  ├── @serialport/parser-cctalk@9.0.1
  ├── @serialport/parser-delimiter@9.0.1
  ├─┬ @serialport/parser-readline@9.0.1
  │ └── @serialport/parser-delimiter@9.0.1 deduped
  ├── @serialport/parser-ready@9.0.1
  ├── @serialport/parser-regex@9.0.1
  ├─┬ @serialport/stream@9.0.1
pi@raspberrypi:~/Desktop/serialTest $ udevadm info -q path -n /dev/ttyUSB0
Unknown device "/dev/ttyUSB0": Permission denied
pi@raspberrypi:~/Desktop/serialTest $ ls /dev | grep USB
ttyUSB0
ttyUSB1
ttyUSB2
ttyUSB3

pi@raspberrypi:~/Desktop/serialTest $ node -v
v14.11.0
pi@raspberrypi:~/Desktop/serialTest $ cat t.js
const SerialPort = require('serialport')

SerialPort.list().then(ports=>console.log(ports), err=>console.log(err))pi@raspberrypi:~/Desktop/serialTest $ node t.js
[]
```

The only real difference on my end seems to be a newer version of node and the fact that `udevadm` gives a permissions error. I added the `ls /dev | grep USB` to show you that the cables are truly there and recognized. I don't know why the udevadm comes back with a permissions error. Maybe I accidentally modified permissions without realizing it? Maybe I should reflash to be sure...

> Is the new 64-bit versions of Raspberry Pi OS being used?

No. I used [Raspberry Pi OS (32-bit) with desktop](https://www.raspberrypi.org/downloads/raspberry-pi-os/)

> How was the image on the SD card created? Was an old version updated or was Raspberry Pi OS August 2020 freshly flashed onto the SD card?

Fresh flash onto a new sd card with latest (August 2020) on the day I opened this issue using [Balena Etcher](https://www.balena.io/etcher/)

> Were permissions modified?

No, at least not intentionally. After first boot installed node, git cloned my working repo, installed serialport (I use yarn) and ran the test script above.

> Are new udev rules needed for the serial cable being used?

Maybe? I'm not super familiar with udev. However, the fact that the .list() function of py-serial and serialport's other read/write functions work suggest no further configuration is need for my cables..?

fivdi, 2020-09-21:
What does `sudo udevadm info -q path -n /dev/ttyUSB0` display?

mikelyndersOKCC, 2020-09-21:
That works as expected:

```console
pi@raspberrypi:~ $ sudo udevadm info -q path -n /dev/ttyUSB0
/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.2/1-1.2.2:1.0/ttyUSB0/tty/ttyUSB0
```

fivdi, 2020-09-22:
I don't really know why it's not working other than that it's a permission issue. Does the file /etc/udev/rules.d/99-com.rules contain the following:
```
SUBSYSTEM=="input", GROUP="input", MODE="0660"
SUBSYSTEM=="i2c-dev", GROUP="i2c", MODE="0660"
SUBSYSTEM=="spidev", GROUP="spi", MODE="0660"
SUBSYSTEM=="bcm2835-gpiomem", GROUP="gpio", MODE="0660"
SUBSYSTEM=="rpivid-*", GROUP="video", MODE="0660"

KERNEL=="vcsm-cma", GROUP="video", MODE="0660"
SUBSYSTEM=="dma_heap", GROUP="video", MODE="0660"

SUBSYSTEM=="gpio", GROUP="gpio", MODE="0660"
SUBSYSTEM=="gpio*", PROGRAM="/bin/sh -c '\
	chown -R root:gpio /sys/class/gpio && chmod -R 770 /sys/class/gpio;\
	chown -R root:gpio /sys/devices/virtual/gpio && chmod -R 770 /sys/devices/virtual/gpio;\
	chown -R root:gpio /sys$devpath && chmod -R 770 /sys$devpath\
'"

SUBSYSTEM=="pwm*", PROGRAM="/bin/sh -c '\
	chown -R root:gpio /sys/class/pwm && chmod -R 770 /sys/class/pwm;\
	chown -R root:gpio /sys/devices/platform/soc/*.pwm/pwm/pwmchip* && chmod -R 770 /sys/devices/platform/soc/*.pwm/pwm/pwmchip*\
'"

KERNEL=="ttyAMA0", PROGRAM="/bin/sh -c '\
	ALIASES=/proc/device-tree/aliases; \
	if cmp -s $ALIASES/uart0 $ALIASES/serial0; then \
		echo 0;\
	elif cmp -s $ALIASES/uart0 $ALIASES/serial1; then \
		echo 1; \
	else \
		exit 1; \
	fi\
'", SYMLINK+="serial%c"

KERNEL=="ttyAMA1", PROGRAM="/bin/sh -c '\
	ALIASES=/proc/device-tree/aliases; \
	if [ -e /dev/ttyAMA0 ]; then \
		exit 1; \
	elif cmp -s $ALIASES/uart0 $ALIASES/serial0; then \
		echo 0;\
	elif cmp -s $ALIASES/uart0 $ALIASES/serial1; then \
		echo 1; \
	else \
		exit 1; \
	fi\
'", SYMLINK+="serial%c"

KERNEL=="ttyS0", PROGRAM="/bin/sh -c '\
	ALIASES=/proc/device-tree/aliases; \
	if cmp -s $ALIASES/uart1 $ALIASES/serial0; then \
		echo 0; \
	elif cmp -s $ALIASES/uart1 $ALIASES/serial1; then \
		echo 1; \
	else \
		exit 1; \
	fi \
'", SYMLINK+="serial%c"
```

mikelyndersOKCC, 2020-09-23:
Mine is identical

I will try to do a fresh install when I have a chance. Another side note is these ports do not show up in CuteCom either. How strange. 

P.S. Could the fact that these ports are plugged into a USB Hub have any effect on this?

```console
pi@raspberrypi:~ $ cat /etc/udev/rules.d/99-com.rules
SUBSYSTEM=="input", GROUP="input", MODE="0660"
SUBSYSTEM=="i2c-dev", GROUP="i2c", MODE="0660"
SUBSYSTEM=="spidev", GROUP="spi", MODE="0660"
SUBSYSTEM=="bcm2835-gpiomem", GROUP="gpio", MODE="0660"
SUBSYSTEM=="rpivid-*", GROUP="video", MODE="0660"

KERNEL=="vcsm-cma", GROUP="video", MODE="0660"
SUBSYSTEM=="dma_heap", GROUP="video", MODE="0660"

SUBSYSTEM=="gpio", GROUP="gpio", MODE="0660"
SUBSYSTEM=="gpio*", PROGRAM="/bin/sh -c '\
	chown -R root:gpio /sys/class/gpio && chmod -R 770 /sys/class/gpio;\
	chown -R root:gpio /sys/devices/virtual/gpio && chmod -R 770 /sys/devices/virtual/gpio;\
	chown -R root:gpio /sys$devpath && chmod -R 770 /sys$devpath\
'"

SUBSYSTEM=="pwm*", PROGRAM="/bin/sh -c '\
	chown -R root:gpio /sys/class/pwm && chmod -R 770 /sys/class/pwm;\
	chown -R root:gpio /sys/devices/platform/soc/*.pwm/pwm/pwmchip* && chmod -R 770 /sys/devices/platform/soc/*.pwm/pwm/pwmchip*\
'"

KERNEL=="ttyAMA0", PROGRAM="/bin/sh -c '\
	ALIASES=/proc/device-tree/aliases; \
	if cmp -s $ALIASES/uart0 $ALIASES/serial0; then \
		echo 0;\
	elif cmp -s $ALIASES/uart0 $ALIASES/serial1; then \
		echo 1; \
	else \
		exit 1; \
	fi\
'", SYMLINK+="serial%c"

KERNEL=="ttyAMA1", PROGRAM="/bin/sh -c '\
	ALIASES=/proc/device-tree/aliases; \
	if [ -e /dev/ttyAMA0 ]; then \
		exit 1; \
	elif cmp -s $ALIASES/uart0 $ALIASES/serial0; then \
		echo 0;\
	elif cmp -s $ALIASES/uart0 $ALIASES/serial1; then \
		echo 1; \
	else \
		exit 1; \
	fi\
'", SYMLINK+="serial%c"

KERNEL=="ttyS0", PROGRAM="/bin/sh -c '\
	ALIASES=/proc/device-tree/aliases; \
	if cmp -s $ALIASES/uart1 $ALIASES/serial0; then \
		echo 0; \
	elif cmp -s $ALIASES/uart1 $ALIASES/serial1; then \
		echo 1; \
	else \
		exit 1; \
	fi \
'", SYMLINK+="serial%c"
```

fivdi, 2020-09-23:
> Could the fact that these ports are plugged into a USB Hub have any effect on this?

Try pluging some of them directly into the Pi USB ports to see if it makes a difference.

fivdi, 2020-09-23:
And remove the USB-Hub for the test.

steuck13, 2020-09-28:
> The code shown above will list the available ports if run by user `pi` on Raspberry Pi OS. If a different user runs the script, that user will need to be a member of group `gpio` in order for the [...]

Is this a new thing? On Raspbian Stretch I have this node working with only 'dialout' group added to my user.

fivdi, 2020-09-28:
I've only noticed this with the current version of Raspberry Pi OS which is based on [Debian](https://www.debian.org/releases/) Buster. For the `list` method to function correctly and return entries the user needs to be a member of group `gpio`. For the other methods to function correctly the user needs to be a member of group `dialout`.

Xeobass, 2020-12-16:
HI !

i have the same problem, but i think is something wrong with OS or hardware.
If i use a RPI 3, my program works like a charm. If i change to RPI4 i only can access to USBx running my program as sudo

And not only with serial communication. I've tryed to make my app autostart (is an electron app) and it worked perfect on RPI3 but it not worked on RPI4.
My raspbian is installed with raspberry pi imager and is the last full raspbian version available (tryed with all versions).
It works with a simple udev rules on ubuntu (RPI3 and RPI4)

If someone have a clue about what's happening....please share it!! XD

easytarget, 2021-03-24:
I've had the same issue on both an RPi3 and RPi4 running `buster`, trying to setup fresh test installs of a server/service I maintain.

In both cases the user was in `dialout` but not in `gpio`, and an empty list was being returned. Adding my user to the 'gpio' group *and starting a new login session* solved this, giving the expected list.
- I can now add this to the existing documentation / set-up instructions for the service on the Pi, thanks for the clear analysis above. ;-)

As a suggestion; if `listLinux()` detects a totally empty list returned by `udevadm info -e` rather than an error, maybe it should flag this as a potential permissions issue. Udevadm should return *some* output in response to the info verb, even if no serial devices are present there should still be a lot of other devices listed, a totally empty response could trigger a 'check permissions' error, or warning, or whatever..

rbgorbet, 2021-06-06:
@mikelyndersOKCC did you ever get a resolution to this, other than the workarounds of hard-coding the port or using OS command like ls `/dev | grep USB` ?  I'm having exactly the same problem in a system where I want a user to serve a `node` webpage that provides a public interface to a serial device, and I really don't want them in `sudo` group.

Since the thread went stale, I wondered if you had any resolution.

rbgorbet, 2021-06-06:
Okay.  For posterity, it seems to be enough for the user to be part of `gpio` to list the ports, and `dialout` to access them, as suggested early on in this thread.  It didn't work for me initially because I didn't restart a shell after being added to `gpio`.  I think this issue should probably be closed, but it's not mine to close.

GazHank, 2021-06-07:
Hi @rbgorbet

It sounds like you might need to configure the admin rules for your device(s).

For one of my projects we use:
`echo 'SUBSYSTEMS=="usb", ATTRS{idVendor}=="1209", ATTRS{idProduct}=="2201", GROUP="users", MODE="0666"' > /etc/udev/rules.d/50-dygma.rules && udevadm control --reload-rules && udevadm trigger
`

This sets the admin rights based on the vendor and product id of the USB device; of course the name of the rules file and the id for the vendor and the product would need to be changed for your usage.  Provided you are seeking to connect to a finite number of specific devices then this type of approach would seem preferable to running as sudo.

I'm not very familiar with udev rules, but this works for my usage.  If you have any issues then let me know and I'll try to help

Cheers

Gareth

rbgorbet, 2021-06-07:
Thanks @GazHank !  In our case I think it was the missing `gpio` membership.  I tried it and it didn't work, but then on reboot I believe that solved the issue.  So to be able to access ports they need to be in `dialout` but if they also need to list them they need to be in `gpio`.  After that we had no issues even though the user isn't in `sudo`.

GazHank, 2021-06-07:
> Thanks @GazHank !  In our case I think it was the missing `gpio` membership.  I tried it and it didn't work, but then on reboot I believe that solved the issue.  So to be able to access ports they [...]

Ah, the old "switch it off, and on again"!  I hate it when a reboot is needed ;-)

So glad you solved this :-D

GazHank, 2021-06-07:
@rbgorbet if there is anything that you think could be added to the documentation to help others with similar issues or troubleshooting then do propose some changes or let me know and I'll try to make some changes on your behalf

closed 2021-07-05 by GazHank