serialport/node-serialport #2143: disable DTR error [Error: Error: Bad file descriptor, cannot set]

lukywong, 2020-09-22:
### Summary of Problem

I'm trying to disable DTR but got error
```
[Error: Error: Bad file descriptor, cannot set] 
```
### Code to Reproduce the Issue

```js
const options = {
  hupcl: false,     
  dataBits: 8, 
  stopbits: 1, 
  parity: 'none' 
};
const sPort = new Serialport(serialport, options);
sPort.on('open', () => {
  sPort.update({ baudRate: 1200 });
  sPort.set({ dtr: false });
  sPort.close();
  console.log(`Forcing using 1200bps open on port "${serialport}"`);
});
```

### Versions, Operating System and Hardware

- SerialPort@9.0.1
- Node.js v12.18.3
- MacOS 10.15.6
- Arduino hardware device

fivdi, 2020-09-25:
`SeriapPort#update`, `SerialPort#set` and `SerialPort#close` are all asynchronous methods. I'm not sure if it's a good idea to call `SerialPort#close` while `SeriapPort#update` and `SerialPort#set` are still running asynchronously. Maybe it's better to wait for `SeriapPort#update` and `SerialPort#set` to complete execution before calling `SerialPort#close`. It doesn't look the best but something like this may work:
```js
const options = {
  hupcl: false,
  dataBits: 8,
  stopbits: 1,
  parity: 'none'
};
const sPort = new Serialport(serialport, options);
sPort.on('open', () => {
  sPort.update({ baudRate: 1200 }, (err) => {
    if (err) throw err;
    sPort.set({ dtr: false }, (err) => {
      if (err) throw err;
      sPort.close((err) => {
        if (err) throw err;
        console.log(`Forcing using 1200bps open on port "${serialport}"`);
      });
    })
  })
});
```

lukywong, 2020-10-01:
Hi @fivdi , thank you, finally I found a simple way to reset serial port by following code, disable dtr is not necessary:

```js
const options = { baudRate: 1200, hupcl: false };
const sPort = new Serialport(serialport, options);
sPort.on('open', () => {
  send({ msg: `Forcing using 1200bps open on port "${serialport}"` });
  sPort.close();
});
sPort.on('close', () => {
  send({ msg: `Forcing using 1200bps close on port "${serialport}"` });
});
```

closed 2020-10-01 by lukywong