serialport/node-serialport #2145: Failing to build Node SerialPort on KDE Neon Linux

leni1, 2020-09-30:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?
Rebuild `serialport` from source

- What happens?
I get the following traceback:
```
user@localhost:~/home/user/MyProject/app$ npm rebuild serialport --build-from-source

> serialport@6.2.2 install /home/user/MyProject/app/node_modules/serialport
> prebuild-install || node-gyp rebuild

make: Entering directory '/home/user/MyProject/app/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
In file included from ../src/./serialport.h:6,
                 from ../src/serialport.cpp:1:
../../nan/nan.h: In function 'void Nan::AsyncQueueWorker(Nan::AsyncWorker*)':
../../nan/nan.h:2294:62: warning: cast between incompatible function types from 'void (*)(uv_work_t*)' {aka 'void (*)(uv_work_s*)'} to 'uv_after_work_cb' {aka 'void (*)(uv_work_s*, int)'} [-Wcast-function-type]
 2294 |     , reinterpret_cast<uv_after_work_cb>(AsyncExecuteComplete)
      |                                                              ^
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Open(Nan::NAN_METHOD_ARGS_TYPE)':
../src/serialport.cpp:41:49: warning: 'v8::String::Utf8Value::Utf8Value(v8::Local<v8::Value>)' is deprecated: Use Isolate version [-Wdeprecated-declarations]
   41 |   v8::String::Utf8Value path(info[0]->ToString());
      |                                                 ^
In file included from /home/user/.cache/node-gyp/10.20.1/include/node/v8.h:26,
                 from /home/user/.cache/node-gyp/10.20.1/include/node/node.h:63,
                 from ../../nan/nan.h:56,
                 from ../src/./serialport.h:6,
                 from ../src/serialport.cpp:1:
/home/user/.cache/node-gyp/10.20.1/include/node/v8.h:2892:28: note: declared here
 2892 |                   explicit Utf8Value(Local<v8::Value> obj));
      |                            ^~~~~~~~~
/home/user/.cache/node-gyp/10.20.1/include/node/v8config.h:324:3: note: in definition of macro 'V8_DEPRECATED'
  324 |   declarator __attribute__((deprecated(message)))
      |   ^~~~~~~~~~
../src/serialport.cpp:48:53: warning: 'v8::Local<v8::Object> v8::Value::ToObject() const' is deprecated: Use maybe version [-Wdeprecated-declarations]
   48 |   v8::Local<v8::Object> options = info[1]->ToObject();
      |                                                     ^
In file included from /home/user/.cache/node-gyp/10.20.1/include/node/node.h:63,
                 from ../../nan/nan.h:56,
                 from ../src/./serialport.h:6,
                 from ../src/serialport.cpp:1:
/home/user/.cache/node-gyp/10.20.1/include/node/v8.h:10046:15: note: declared here
10046 | Local<Object> Value::ToObject() const {
      |               ^~~~~
../src/serialport.cpp:78:69: warning: cast between incompatible function types from 'void (*)(uv_work_t*)' {aka 'void (*)(uv_work_s*)'} to 'uv_after_work_cb' {aka 'void (*)(uv_work_s*, int)'} [-Wcast-function-type]
   78 |   uv_queue_work(uv_default_loop(), req, EIO_Open, (uv_after_work_cb)EIO_AfterOpen);
      |                                                                     ^~~~~~~~~~~~~
../src/serialport.cpp: In function 'void EIO_AfterOpen(uv_work_t*)':
../src/serialport.cpp:95:30: warning: 'v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const' is deprecated [-Wdeprecated-declarations]
   95 |   data->callback.Call(2, argv);
      |                              ^
In file included from ../src/./serialport.h:6,
                 from ../src/serialport.cpp:1:
../../nan/nan.h:1742:3: note: declared here
 1742 |   Call(int argc, v8::Local<v8::Value> argv[]) const {
      |   ^~~~
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Update(Nan::NAN_METHOD_ARGS_TYPE)':
../src/serialport.cpp:113:53: warning: 'v8::Local<v8::Object> v8::Value::ToObject() const' is deprecated: Use maybe version [-Wdeprecated-declarations]
  113 |   v8::Local<v8::Object> options = info[1]->ToObject();
      |                                                     ^
In file included from /home/user/.cache/node-gyp/10.20.1/include/node/node.h:63,
                 from ../../nan/nan.h:56,
                 from ../src/./serialport.h:6,
                 from ../src/serialport.cpp:1:
/home/user/.cache/node-gyp/10.20.1/include/node/v8.h:10046:15: note: declared here
10046 | Local<Object> Value::ToObject() const {
[... 193 lines cut ...]
/home/user/.cache/node-gyp/10.20.1/include/node/node_object_wrap.h:84:78:   required from here
/home/user/.cache/node-gyp/10.20.1/include/node/v8.h:9502:16: warning: cast between incompatible function types from 'v8::WeakCallbackInfo<node::ObjectWrap>::Callback' {aka 'void (*)(const v8::WeakCallbackInfo<node::ObjectWrap>&)'} to 'Callback' {aka 'void (*)(const v8::WeakCallbackInfo<void>&)'} [-Wcast-function-type]
 9502 |                reinterpret_cast<Callback>(callback), type);
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/user/.cache/node-gyp/10.20.1/include/node/v8.h: In instantiation of 'void v8::PersistentBase<T>::SetWeak(P*, typename v8::WeakCallbackInfo<P>::Callback, v8::WeakCallbackType) [with P = Nan::ObjectWrap; T = v8::Object; typename v8::WeakCallbackInfo<P>::Callback = void (*)(const v8::WeakCallbackInfo<Nan::ObjectWrap>&)]':
../../nan/nan_object_wrap.h:65:61:   required from here
/home/user/.cache/node-gyp/10.20.1/include/node/v8.h:9502:16: warning: cast between incompatible function types from 'v8::WeakCallbackInfo<Nan::ObjectWrap>::Callback' {aka 'void (*)(const v8::WeakCallbackInfo<Nan::ObjectWrap>&)'} to 'Callback' {aka 'void (*)(const v8::WeakCallbackInfo<void>&)'} [-Wcast-function-type]
  CXX(target) Release/obj.target/serialport/src/serialport_linux.o
  SOLINK_MODULE(target) Release/obj.target/serialport.node
  COPY Release/serialport.node
make: Leaving directory '/home/user/MyProject/app/node_modules/serialport/build'
serialport@6.2.2 /home/user/MyProject/app/node_modules/serialport
```
- What should have happened?
A successful build of `serialport`

### Code to Reproduce the Issue
`npm rebuild serialport --build-from-source`

### Versions, Operating System and Hardware

- SerialPort@6.2.2
- Node.js v10.20.1
- Linux: [KDE Neon](https://neon.kde.org/) 5.19.5
- Hardware and chipset: Intel i7-8550U CPU

fivdi, 2020-09-30:
The messages that are being displayed are warnings rather than errors and shouldn't result in any problems. It looks like serialport was built successfully and I would expect it to function correctly. Please give it a try to see if it works.

leni1, 2020-10-01:
Hi. Thanks for letting me know. Closing this since `serialport` seems to be working. The problem may be coming from somewhere else. 😃

closed 2020-10-01 by leni1