serialport/node-serialport #2146: Read data with weighing machine connected through USB to serial interface

fdbatista, 2020-10-01:
### What are you trying to do?
I want to read data from a digital weighing machine connected via USB to port COM4 on my Windows 10 computer.
To do so, I need to send the character 'P' to the weighing machine and then I need to read the response.
I was trying to send the data as hex (0x50)  but it throws an error.

### What happens?
Nothing. The port gets correctly opened, but it looks like there is no communication with it. The error and data variables from the callback always show `undefined`.

### What should have happened?
I was expecting to be able to send the character 'P' to the COM port and get a floating point number as a result.

### Code to Reproduce the Issue

```js
const SerialPort = require('serialport')

try {
    var com = new SerialPort('COM4',{
        autoOpen:false,
        baudRate: 115200,
        databits: 8,
        parity: 'none'
    });

    com.open(function (error) {
        if (error) {
            console.log('Error while opening the port ' + error);
        } else {
            console.log('Port open');

            var sentData = 'P'

            try {
                com.write(sentData, function (err, result) {
                    console.log(err);
                    console.log(result);
                });
            } catch (error) {
                console.log('Error');
                console.log(error);
            }

            com.close();
            console.log('Port closed')

        }
    });
} catch (error) {
    console.log(error);
}
```

### Versions, Operating System and Hardware

- SerialPort@9.0.1
- Node.js v14.12.0
- Windows 10 Home Edition
- Hardware and chipset: Torrey PCR Commercial Weighing Machine

jsftsylvain, 2020-10-01:
I am having an issue where I cannot read data from the serialport when I declare it as a variable. Try declaring it as a constant.
[my issue](https://github.com/serialport/node-serialport/issues/2136)

fdbatista, 2020-10-01:
> I am having an issue where I cannot read data from the serialport when I declare it as a variable. Try declaring it as a constant.
> [my issue](https://github.com/serialport/node-serialport/issues/2136)

Thanks, @jsftsylvain I will give it a try and let you know what happens. Cheers!

fdbatista, 2020-10-03:
> I am having an issue where I cannot read data from the serialport when I declare it as a variable. Try declaring it as a constant.
> [my issue](https://github.com/serialport/node-serialport/issues/2136)

Him @jsftsylvain You were right, declaring it as a `const` did the job.
I managed to solve this issue by subscribing to the port events outside of the scope of the `open` listener.

And I ended up creating my own wrapper for this job.
You can check it out here:

https://github.com/fdbatista/nodejs-torrey-weighing-machine

It is still work in progress, but any suggestions are welcome 👍 

Thanks.

closed 2020-12-19 by stale[bot]