serialport/node-serialport #2148: How to implement port.readline()?

heltonbiker, 2020-10-02:
### Synchronous readline: how to do it?

- What are you trying to do?

I would like to call something like `response = port.readline();` instead of using the event-driven `ReadLine` parser, or polling the port with "port.read()".
My device implements a protocol where it expects that I only send a new message after having received the response from the previous message sent.

- What happens?

If I use readline parser and send multiple writes, it is cumbersome to match the "writes" with the respective "responses", since the writes are sent all at once, and the responses arrive shortly after (at least according to my console logs). Sample code below:

    const SerialPort = require('serialport');
    const ReadLine = require('@serialport/parser-readline');
    const api = require('./myApi');

    SerialPort.list()
      .then(portInfos => {
        portInfos.filter(pinfo => pinfo.manufacturer === 'FTDI')
          .forEach(async portInfo => {
            const port = new SerialPort(portInfo.path, {
              baudRate: 3000000,
              parity: 'even',
              dataBits: 7,
              rtscts: true,
              autoOpen: false,
            }).setEncoding('ascii');

            const parser = port.pipe(new ReadLine({
              delimiter: '\r\n',
              encoding: 'ascii',
            }));
            parser.on('data', data => {
              console.log('Rx', data);
              console.log();
            });

            port.open();

            const serialMessage = api.createReadMessage(api.SERIAL);
            const batteryMessage = api.createReadMessage(api.BATTERY);

            for (const message of [serialMessage, batteryMessage]) {
              port.write(message);
              console.log('Tx', message.trim());
            }
          });
      });

Output:

> Tx :0A0300070004E8
> Tx :0A03000B0002E6
> Rx :0A0308000004670000000A76
>
> Rx :0A03040064006427
>

- What should have happened?

If I could devise a function like

    function sendAndReceive(message, port) {
        port.write(message);
        response = port.readline();
        console.log('Tx', message);
        console.log('Rx, response);
        console.log();
        return response();
    }

Then I could have my code like this:

    batteryResponse = sendAndReceive(batteryRequestMessage, port);
    serialNumberResponse = sendAndReceive(serialNumberRequestMessage, port);

and get this output:

> Tx :0A0300070004E8
> Rx :0A0308000004670000000A76
>
> Tx :0A03000B0002E6
> Rx :0A03040064006427
> 

### Versions, Operating System and Hardware

- SerialPort@9.0.0
- Node.js v10.15.3
- Windows 10
- FTDI (custom device)

heltonbiker, 2020-10-02:
Well I came up with this, which is not a proper "readLine" method, but seems to work for a possible "sendAndReceive" implementation. I am not sure if this is reliable, though. Sorry if this is not the right place to ask, but any help is appreciated.

    const sendAndReceive = (message, port) => new Promise((resolve) => {
      port.write(message);
      const parser = port.pipe(new ReadLine({
        delimiter: '\r\n',
        encoding: 'ascii',
      }));
      parser.on('data', data => {
        parser.removeAllListeners('data');
        resolve(data);
      });
    });

closed 2020-12-19 by stale[bot]

fromage9747, 2024-04-22:
> Well I came up with this, which is not a proper "readLine" method, but seems to work for a possible "sendAndReceive" implementation. I am not sure if this is reliable, though. Sorry if this is not [...]

Thanks for this.  On initial testing, it looks good.  I will continue testing it, and hopefully it will solve my issue especially when running every 1 to 10 milliseconds.

Cheers

fromage9747, 2024-04-22:
Unfortunately, it works great until you hit it hard.  Responses get mixed up when running fast.