serialport/node-serialport #2151: Can not return List of available COM ports

Shanmax1996, 2020-10-10:
I'm having a problem with a part of my program and I think I know what the problem is I can't find a way to fix it, hence need your help

This my code. Here I am trying to return the comport as a variable.

```
const SerialPort = require('serialport')

var getPortsList = ()=>{
  var portsList = [];
  
  SerialPort.list().then(ports =>{
    ports.forEach((ports)=>{
      portsList.push(ports.path);
      console.log(portsList);
    });
  });

  return portsList;

};

console.log(getPortsList());
```

output

```
node test
[]
[ 'COM3' ]
```

I got an empty array of list
Please help me Thank you

reconbot, 2020-10-10:
Since the results are async you need to use the callback or return a promise.

Shanmax1996, 2020-10-11:
I also used that way but it doesn't work. I am trying to figure out my problem but I don't
Please help me.

```
var getPortsList = (callback) => {
  var portsList = [];

  SerialPort.list().then(ports => {
    ports.forEach((port) => {
      portsList.push(port.path);
    });
    
    callback(null, portsList);
  });
};

console.log(getPortsList());
```

Shanmax1996, 2020-10-11:
I solve that way. Thanks for your supporting 😍

```
var SerialPort = require('serialport');
let arduinoSerialPort;

(async () => {
  try {
    const serialList = await SerialPort.list();
    const path = serialList[0].path;
    console.log('openning port :', path );
    arduinoSerialPort = new SerialPort(path, {baudRate: 115200});
  } catch (e) {
    console.log(e);
  }
})();
```

closed 2020-10-11 by Shanmax1996