serialport/node-serialport #2156: Node ABI issue with 9.0.2 bindings

choppsta, 2020-10-20:
### Summary of Problem

- What are you trying to do?

Install serialport with prebuilt bindings

- What happens?

They cannot be found

- What should have happened?

They should be matched against the node ABI number and downloaded

### Versions, Operating System and Hardware

- SerialPort@9.0.2
- Node.js 12+
- Windows, Linux, Mac

It looks like the process that builds the prebuilt bindings is affected by this issue:
https://github.com/lgeiger/node-abi/issues/90

The filenames of the bindings don't seem to have the right ABI numbers (e.g. node 12 should be 72, but I believe it's incorrectly being labelled as 68).

It would also be really great to get prebuilt bindings for electron 10 (ABI 82)!

reconbot, 2020-11-03:
Until that issue closes there's not much we can do.

Lovinity, 2020-11-20:
It looks like node-abi released an update a few days ago.

jeffrson, 2020-12-04:
Also there are'nt binary files for node 14 (LTS!) ABI 83.

reconbot, 2020-12-04:
Try `serialport@9.0.3`

reconbot, 2020-12-04:
And as always I welcome help maintaining the project, I'm not able to attend to these changes promptly.

choppsta, 2020-12-04:
Thanks reconbot, unfortunately I think there is still an issue with the bindings build process as the ABI numbers for the node bindings don't seem to match (https://github.com/lgeiger/node-abi/blob/master/abi_registry.json).  The electron ones are ok, although they only go up to 80 (Electron 9) where as the current version of Electron is two major versions ahead now.  A lot of the recent issues on this repo seem to be related to this.

I really appreciate all the work you put into this project and would love to help out if I can.  Could you point me in the right direction for how the building of these bindings works?

choppsta, 2020-12-04:
Hi reconbot, I've dug into this and found the source of the issue:

The version of the `node-abi` in the `package-lock.json` is out of date.  It's version `2.19.1` when the current version is `2.19.3`.  Because it's a dependency of `prebuild` it won't get updated unless that project specifies a higher version.

Quick solution is `npm update --depth 2 node-abi`, although it might be better to add `node-abi` as a dependency to this project meaning that it could be kept up to date more easily with `npm update`.

Hope this helps.  Let me know I can do anything to help further.

jeffrson, 2020-12-07:
Good find @choppsta!

Although the "bindings" workspace has an updated `package-lock.json`, `prebuild` is installed from the workspace's root and so it uses the `node-abi` that's available there.

I can confirm, that with @choppsta's "quick fix" correctly versioned packages (ABI) can be built.

NoahAndrews, 2020-12-16:
@choppsta Ah, that's annoying. I actually added `node-abi` as an explicit dependency for the bindings package, but I didn't realize it needed to be done at the top-level as well. I'll make another PR.

reconbot, 2020-12-17:
Released in 9.0.4, binaries should be available int he next 30-45 minutes

reconbot, 2020-12-17:
Looks like we're having build issues on linux and windows. I'm digging into it.

reconbot, 2020-12-17:
I've ported the builds to github actions, seems to be in good shape.

choppsta, 2020-12-18:
That's great, thanks everyone!  I can confirm this now works perfectly, even for the latest electron version (11.1.0).

closed 2020-12-18 by choppsta

Lovinity, 2020-12-18:
I must be doing something wrong because I'm still getting bindings 9.0.4 which has the Windows/Linux build error that @reconbot mentioned. 9.0.5 seems to be unpublished.

reconbot, 2020-12-18:
There is no 9.0.5, I got it working for 9.0.4