serialport/node-serialport #2158: SerialPort is not compatible with yarn2

gjuchault, 2020-10-21:
### Summary of Problem

- What are you trying to do? I'm trying to use node-serialport with yarn 2 (berry)
- What happens? I get the following message: `"offset is not an integer got NaN"`
- What should have happened? I should have been able to communicate with the USB device

I migrated my project to yarn berry and I got the following error with RFXCom library `"offset" is not an integer got "NaN"`. Googling the error went to this other node-serialport issue https://github.com/serialport/node-serialport/issues/2014. The fact that it was related to PM2 gave me the hint. I tried without yarn berry and it worked again.

### Versions, Operating System and Hardware

- SerialPort@9.0.1?
- Node.js v15?
- Windows? Linux? Mac? Linux
- Hardware and chipset? (Prolific/FTDI/Other) RFXCom 433Hz

Mike-Dax, 2020-11-13:
I can replicate this bug. Running the test with `DEBUG=serialport/stream` I get this:

```
  serialport/stream .list +0ms
  serialport/stream .list +12ms
  serialport/stream _read queueing _read for after open +14ms
  serialport/stream _read queueing _read for after open +0ms
  serialport/stream opening path: /dev/tty.Bluetooth-Incoming-Port +3ms
  serialport/stream opening path: /dev/tty.usbmodem14601 +0ms
  serialport/stream opened path: /dev/tty.Bluetooth-Incoming-Port +1ms
  serialport/stream _read reading { start: 0, toRead: 65536 } +0ms
  serialport/stream #set { brk: false, cts: false, dtr: true, dts: false, rts: false } +1ms
  serialport/stream #close +1ms
  serialport/stream binding.set finished +0ms
  serialport/stream opened path: /dev/tty.usbmodem14601 +0ms
  serialport/stream _read reading { start: 0, toRead: 65536 } +0ms
  serialport/stream #set { brk: false, cts: false, dtr: true, dts: false, rts: false } +1ms
  serialport/stream binding.close finished +7ms
  serialport/stream binding.read error [Error: EBADF: bad file descriptor, read] {
  errno: -9,
  code: 'EBADF',
  syscall: 'read',
  disconnect: true
} +1ms
  serialport/stream disconnected aborted because already closed [Error: EBADF: bad file descriptor, read] {
  errno: -9,
  code: 'EBADF',
  syscall: 'read',
  disconnect: true
} +0ms
  serialport/stream _read queueing _read for after open +1ms
  serialport/stream _write 9 bytes of data +3ms
  serialport/stream drain +0ms
  serialport/stream binding.set finished +1ms
  serialport/stream binding.write write finished +0ms
  serialport/stream binding.drain finished +0ms
  serialport/stream binding.read finished { bytesRead: undefined } +0ms
  serialport/stream _read reading { start: NaN, toRead: NaN } +1ms
```

It seems the root cause is something causing `[Error: EBADF: bad file descriptor, read]`, but that's as far as I've gone so far.

Mike-Dax, 2020-11-17:
This has been fixed upstream. I've tested the fix locally and I'm able to run serialport within a yarn v2 environment.

reconbot, 2020-11-17:
That’s fantastic

closed 2020-11-20 by reconbot