serialport/node-serialport #2159: Serialport SHOULD support the latest version of Electron.

timothyjrbeck, 2020-10-24:
## 💥 Proposal

### What feature you'd like to see
(A clear and concise description of what the proposal is.)

Support for the latest version of Electron.

## Motivation
(Please outline the motivation for the proposal. It's interesting knowing what people are working on and also could help community members make suggestions for work-arounds until the feature is built)

As per the error message below, It seems like it's impossible to use the latest version of Electron (10.x) and serialport at the same time.

> Error: The module '\node_modules\@serialport\bindings\build\Release\bindings.node'
> was compiled against a different Node.js version using
> NODE_MODULE_VERSION 64. This version of Node.js requires
> NODE_MODULE_VERSION 84. Please try re-compiling or re-installing

Not using the latest version might cause some security issues, hence I think this problem should be addressed as soon as possible.

reconbot, 2020-10-24:
Patches welcome, it usually involves updating prebuild.

Lovinity, 2020-11-20:
It is possible to use node-serialport and the latest Electron if you build from source. Unfortunately, node-serialport does not support source building on MacOS (at least, that's the error I get). I would love to at the very least see support for source building on MacOS.

It is very critical to support later versions of Electron soon. Given the latest binary-available version is electron v9, once electron v12 is out, electron v9 will no longer be supported by the Electron team.

reconbot, 2020-11-21:
It absolutely does support building locally.

Lovinity, 2020-11-21:
That's odd. When I tried to compile on OSX 10.2 it said building wasn't supported on that architecture.

reconbot, 2020-11-21:
[Github uses](https://github.com/actions/virtual-environments/blob/main/images/macos/macos-11.0-Readme.md)
> System Version: macOS 11.0.1 (20B29)

And I use 
> 10.15.7

but nothing I know about has changed substantially since v8 iirc

Lovinity, 2020-11-21:
I'll see if I can bump the build server up to macOS 11 and see if that resolves the problem

Lovinity, 2020-11-21:
Okay, I got it backwards. MacOS was the only architecture that supported compiling from source. It failed both on linux and on win32.

The most likely reason for this is the build environment itself runs in MacOS... it has to because you cannot compile anything MacOS on another architecture because of their proprietary XCode lockdown. I am going to try and split the build environment into multiple architectures and see if that fixes the issue.

reconbot, 2020-12-04:
It takes time to maintain, test and upgrade. Electron has a paid team of many people working on it, we're always going to be behind. The only way to fix that is to volunteer. Like @NoahAndrews who PR'd the change that should have fixed this. Or the few people who have attempted N-API ports.

try `serialport@9.0.3`

Lovinity, 2020-12-16:
Understood. I do not have enough knowledge to contribute though, so I am forced to deprecate functionality given electron 9 is about to end support, and even 9.0.3 still does not support the latest Electron ABI (which is 85 for 11.1.0). It only goes up to 80 (which is 9.3.x).

reconbot, 2020-12-17:
Released in 9.0.4, binaries should be available int he next 30-45 minutes, might work for you.

reconbot, 2020-12-17:
Looks like we're having build issues on linux and windows. I'm digging into it.

reconbot, 2020-12-17:
I've ported the builds to github actions, seems to be in good shape.

Sourabh-Zutti, 2021-03-06:
I am not able to find proper bindings for serialport with electron version > 4.
After running electron-rebuild I am getting this below error.
Error Message :

..\src\serialport.cpp(329): error C2661: 'v8::Object::Set': no overloaded function takes 2 arguments [E:\LastestUpdated
Workspace\workspace\profitguru\profitGuruDAPP\node_modules@serialport\bindings\build\bindings.vcxproj]
..\src\serialport.cpp(330): error C2661: 'v8::Object::Set': no overloaded function takes 2 arguments [E:\LastestUpdated
Workspace\workspace\profitguru\profitGuruDAPP\node_modules@serialport\bindings\build\bindings.vcxproj]
..\src\serialport.cpp(331): error C2661: 'v8::Object::Set': no overloaded function takes 2 arguments [E:\LastestUpdated
Workspace\workspace\profitguru\profitGuruDAPP\node_modules@serialport\bindings\build\bindings.vcxproj]
..\src\serialport.cpp(378): error C2661: 'v8::Object::Set': no overloaded function takes 2 arguments [E:\LastestUpdated
Workspace\workspace\profitguru\profitGuruDAPP\node_modules@serialport\bindings\build\bindings.vcxproj]
× Rebuild Failed

I am trying to find the Bindings for serialport.
I am getting errors, not able to get Bindings for serialport
Should get Proper Bindings
Versions, Operating System and Hardware
SerialPort@7.1.5
Node.js v10.15.3
Windows 10
electron@9.2.0

GazHank, 2021-07-05:
This should now be fixed for recent versions of electron.  However would there be value in automating process of updating and rebuilding whenever node-abi is updated to keep things up to date?

GazHank, 2021-07-12:
Dependabot has now been deployed (https://github.com/serialport/node-serialport/pull/2294) so hopefully we can keep on top of future electron updates more quickly.

Since the original issue is resolved, and we have an improved process for the future I'll close this issue.  If anyone has any concerns then please shout!

closed 2021-07-12 by GazHank

reconbot, 2021-07-13:
🙌