serialport/node-serialport #2161: electron npm install serialport error

syanfox, 2020-10-30:
- What are you trying to do?
   npm install serialpor

- What happens?
  Error: Cannot find module 'D:\electron\serialport-app\node_modules\@serialport\bindings\node C:\Users\xxx\AppData\Roaming\npm\node_modules\node-gyp\bin\node-gyp.js'

> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=12.19.0 runtime=node arch=x64 libc= platform=win32)

D:\electron\serialport-app\node_modules\@serialport\bindings>if not defined npm_config_node_gyp (node "C:\Users\xxx\AppData\Roaming\npm\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "node C:\Users\qiushui\AppData\Roaming\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild )
internal/modules/cjs/loader.js:834
  throw err;
  ^

Error: Cannot find module 'D:\electron\serialport-app\node_modules\@serialport\bindings\node C:\Users\xxx\AppData\Roaming\npm\node_modules\node-gyp\bin\node-gyp.js'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:831:15)
    at Function.Module._load (internal/modules/cjs/loader.js:687:27)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:60:12)
    at internal/main/run_main_module.js:17:47 {
  code: 'MODULE_NOT_FOUND',
  requireStack: []
}
npm WARN @ant-design/icons@4.2.2 requires a peer of react@16.x but none is installed. You must install peer dependencies yourself.
npm WARN @ant-design/react-slick@0.27.11 requires a peer of react@^0.14.0 || ^15.0.1 || ^16.0.0 but none is installed. You must install peer dependencies yourself.
npm WARN @ant-design/react-slick@0.27.11 requires a peer of react-dom@^0.14.0 || ^15.0.1 || ^16.0.0 but none is installed. You must install peer dependencies yourself.
npm WARN rc-field-form@1.13.0 requires a peer of react@^16.9.0 but none is installed. You must install peer dependencies yourself.
npm WARN rc-image@3.2.2 requires a peer of react@^16.8.0 but none is installed. You must install peer dependencies yourself.
npm WARN rc-image@3.2.2 requires a peer of react-dom@^16.8.0 but none is installed. You must install peer dependencies yourself.
npm WARN tsutils@3.17.1 requires a peer of typescript@>=2.8.0 || >= 3.2.0-dev || >= 3.3.0-dev || >= 3.4.0-dev || >= 3.5.0-dev || >= 3.6.0-dev || >= 3.6.0-beta || >= 3.7.0-dev || >= 3.7.0-beta but none is installed. You must install peer dependencies yourself.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.1.3 (node_modules\fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.1.3: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.13 (node_modules\watchpack-chokidar2\node_modules\fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.13: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.13 (node_modules\webpack-dev-server\node_modules\fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.13: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@9.0.2 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the @serialport/bindings@9.0.2 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

- SerialPort@?
  9.0.2
- Node.js v?
12.19  -- 
electron  
 10.1.5
- Windows? Linux? Mac?
Windows10 
- Hardware and chipset? (Prolific/FTDI/Other)

syanfox, 2020-11-04:
have already been solved

closed 2020-11-04 by syanfox

stephendpmurphy, 2020-11-05:
@syanfox solved how? Do you mind posting your solution?