serialport/node-serialport #2164: node-serialport for PWA, causes error with bindings on localhost or netlify.

moothyknight, 2020-11-03:
### Summary of Problem

I am trying to get a progressive web app to run the node-serialport package on desktop so I can migrate from chrome.serial. First tried on a localhost then on a hosted netlify webserver with different errors.

### Code to Reproduce the Issue

Initial Test
```js
const serialport = require('serialport');

```

Error on localhost:
```
bindings.js:203 Uncaught TypeError: exists is not a function
    at Function.getRoot (bindings.js:203)
    at bindings (bindings.js:82)
    at Object.parcelRequire.../node_modules/@serialport/bindings/lib/linux.js.util (linux.js:2)
    at newRequire (src.77de5100.js:47)
    at localRequire (src.77de5100.js:53)
    at Object.parcelRequire.../node_modules/@serialport/bindings/lib/index.js.debug (index.js:14)
    at newRequire (src.77de5100.js:47)
    at localRequire (src.77de5100.js:53)
    at Object.parcelRequire.../node_modules/serialport/lib/index.js.@serialport/stream (index.js:2)
    at newRequire (src.77de5100.js:47)
```

Error on netlify (hosted webapp):
```
bindings.js:178 Uncaught TypeError: Cannot read property 'indexOf' of undefined
    at Function.getFileName (bindings.js:178)
    at u (bindings.js:82)
    at Object.<anonymous> (linux.js:2)
    at f (src.690f0ea0.js?__WB_REVISION__=2fa39e153739be0eecc364af075d5850:1)
    at p (src.690f0ea0.js?__WB_REVISION__=2fa39e153739be0eecc364af075d5850:1)
    at Object.<anonymous> (index.js:14)
    at f (src.690f0ea0.js?__WB_REVISION__=2fa39e153739be0eecc364af075d5850:1)
    at p (src.690f0ea0.js?__WB_REVISION__=2fa39e153739be0eecc364af075d5850:1)
    at Object.<anonymous> (nodeserialUtils.js:2)
    at f (src.690f0ea0.js?__WB_REVISION__=2fa39e153739be0eecc364af075d5850:1)
```

 I'm using Parcel as a bundler so it could be a bundling issue.

It appears to be the bindings causing a conflict, as it goes away by requiring @serialport/stream but then I lose the automatic bindings...

Question is, is this actually a viable way to use node-serialport? It should at least work on localhost, right? Or do I absolutely need to wrap it in something like Electron for desktop use?

### Versions, Operating System and Hardware

- SerialPort@9.0.2
- Node.js v14
- Windows

reconbot, 2020-11-03:
We don’t work in browsers.

moothyknight, 2020-11-03:
@reconbot 

Thanks for the quick reply! Okay so I already have a chrome extension set up to serve my purposes, I should just use what I can from the npm packages I like to enhance it rather than trying to rewrite the serial code for nodejs since it looks like it doesn't quite fit for my purposes.

Come on w3 where's muh browser serial ports!

closed 2020-11-03 by moothyknight

moothyknight, 2020-11-07:
@reconbot 

Found my solution:
https://codelabs.developers.google.com/codelabs/web-serial/#1 

Future computing to the rescue!