serialport/node-serialport #2166: Electron-serialport example does not work when following instructions

deenine, 2020-11-08:
### Summary of Problem

(Please answer all 3)

- What are you trying to do? Install and run the electron-serialport example, per https://github.com/serialport/electron-serialport/blob/master/README.md
- What happens? Build error of serialport.o
- What should have happened? succesfull install with no errors

### Code to Reproduce the Issue

```% !rm
rm -rf electron-serialport
% npm --version                                             
6.14.8
% node --version
v14.15.0
% npm cache clean --force                                    
npm WARN using --force I sure hope you know what you are doing.
% git clone https://github.com/serialport/electron-serialport
Cloning into 'electron-serialport'...
remote: Enumerating objects: 256, done.
remote: Total 256 (delta 0), reused 0 (delta 0), pack-reused 256
Receiving objects: 100% (256/256), 61.49 KiB | 224.00 KiB/s, done.
Resolving deltas: 100% (130/130), done.
% cd electron-serialport 
electron-serialport % npm install

> @serialport/bindings@2.0.8 install /Users/deenine/workspace/scratch/electron-serialport/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=14.15.0 runtime=node arch=x64 libc= platform=darwin)
 CXX(target) Release/obj.target/bindings/src/serialport.o
In file included from ../src/serialport.cpp:1:
In file included from ../src/./serialport.h:6:
In file included from ../../../nan/nan.h:2884:
../../../nan/nan_typedarray_contents.h:34:43: warning: 'GetContents' is deprecated: Use GetBackingStore. See http://crbug.com/v8/9908.
     [-Wdeprecated-declarations]
     data   = static_cast<char*>(buffer->GetContents().Data()) + byte_offset;
                                         ^
/Users/deenine/Library/Caches/node-gyp/14.15.0/include/node/v8.h:5265:3: note: 'GetContents' has been explicitly marked deprecated here
 V8_DEPRECATE_SOON("Use GetBackingStore. See http://crbug.com/v8/9908.")
 ^
/Users/deenine/Library/Caches/node-gyp/14.15.0/include/node/v8config.h:402:39: note: expanded from macro 'V8_DEPRECATE_SOON'
# define V8_DEPRECATE_SOON(message) [[deprecated(message)]]
                                     ^
../src/serialport.cpp:329:14: error: no matching member function for call to 'Set'
   results->Set(Nan::New<v8::String>("cts").ToLocalChecked(), Nan::New<v8::Boolean>(data->cts));
   ~~~~~~~~~^~~
/Users/deenine/Library/Caches/node-gyp/14.15.0/include/node/v8.h:3670:37: note: candidate function not viable: requires 3 arguments, but 2 were
     provided
 V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context,
                                   ^
/Users/deenine/Library/Caches/node-gyp/14.15.0/include/node/v8.h:3673:37: note: candidate function not viable: requires 3 arguments, but 2 were
     provided
 V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context, uint32_t index,
                                   ^
../src/serialport.cpp:330:14: error: no matching member function for call to 'Set'
   results->Set(Nan::New<v8::String>("dsr").ToLocalChecked(), Nan::New<v8::Boolean>(data->dsr));
   ~~~~~~~~~^~~
/Users/deenine/Library/Caches/node-gyp/14.15.0/include/node/v8.h:3670:37: note: candidate function not viable: requires 3 arguments, but 2 were
     provided
 V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context,
                                   ^
/Users/deenine/Library/Caches/node-gyp/14.15.0/include/node/v8.h:3673:37: note: candidate function not viable: requires 3 arguments, but 2 were
     provided
 V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context, uint32_t index,
                                   ^
../src/serialport.cpp:331:14: error: no matching member function for call to 'Set'
   results->Set(Nan::New<v8::String>("dcd").ToLocalChecked(), Nan::New<v8::Boolean>(data->dcd));
   ~~~~~~~~~^~~
/Users/deenine/Library/Caches/node-gyp/14.15.0/include/node/v8.h:3670:37: note: candidate function not viable: requires 3 arguments, but 2 were
     provided
 V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context,
                                   ^
/Users/deenine/Library/Caches/node-gyp/14.15.0/include/node/v8.h:3673:37: note: candidate function not viable: requires 3 arguments, but 2 were
     provided
 V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context, uint32_t index,
                                   ^
../src/serialport.cpp:378:14: error: no matching member function for call to 'Set'
   results->Set(Nan::New<v8::String>("baudRate").ToLocalChecked(), Nan::New<v8::Integer>(data->baudRate));
   ~~~~~~~~~^~~
/Users/deenine/Library/Caches/node-gyp/14.15.0/include/node/v8.h:3670:37: note: candidate function not viable: requires 3 arguments, but 2 were
     provided
 V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context,
                                   ^
/Users/deenine/Library/Caches/node-gyp/14.15.0/include/node/v8.h:3673:37: note: candidate function not viable: requires 3 arguments, but 2 were
     provided
 V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context, uint32_t index,
                                   ^
1 warning and 4 errors generated.
make: *** [Release/obj.target/bindings/src/serialport.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)
gyp ERR! System Darwin 19.6.0
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/deenine/workspace/scratch/electron-serialport/node_modules/@serialport/bindings
gyp ERR! node -v v14.15.0
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@2.0.8 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the @serialport/bindings@2.0.8 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/deenine/.npm/_logs/2020-11-08T13_17_12_459Z-debug.log
```

After the build has failed, the following commands will allow this to complete succesfully:
```
npm remove serialport
npm install -build-from-source serialport
npm remove electron
npm install -build-from-source electron
npm start
```

### Versions, Operating System and Hardware

- SerialPort@? does not succesfully install, npm cache cleared, so presumably latest?
- Node.js v? v14.15.0
- Windows? Linux? Mac? OSX 10.15.7
- Hardware and chipset? (Prolific/FTDI/Other) N/A, does not succesfully install.

deenine, 2020-11-08:
Further testing shows that while the electron app loads, the serialport module is not functioning correctly:
```
Uncaught Error: Loading non-context-aware native module in renderer: '/Users/deenine/workspace/scratch/electron-serialport/node_modules/@serialport/bindings/build/Release/bindings.node', but app.allowRendererProcessReuse is true. See https://github.com/electron/electron/issues/18397.
```

Setting `app.allowRendererProcessReuse = false` in main.js removes this warning, however it then fails with:
```
Uncaught (in promise) TypeError: SerialPort.list no longer takes a callback and only returns a promise
    at Function.SerialPort.list (/Users/deenine/workspace/scratch/electron-serialport/node_modules/@serialport/stream/lib/index.js:651)
    at renderer.js:8
```

What is the correct combination of electron and serial port versions to remove these errors?

stephendpmurphy, 2020-11-09:
@deenine I'm currently running the following.

electron: 10.1.4
node: 12.16.3
serialport: 9.0.2

Two things needed to get the example working, one of which you have already done
```
app.allowRendererProcessReuse = false
```

Here is the source I am using for retrieving serial ports.. You could update this yourself in the example to get it running.
```js
var getAvailableSerialPorts = async function() {
            const SerialPort = require('serialport');
            var portList = [];
            try {
                SerialPort.list().then((ports, err) => {
                    if (err) {
                        console.log("Error: ", err.message);
                    }

                    if (ports.length !== 0) {
                        // Iterate through each port and push it into our list if it
                        // meets our criteria
                        ports.forEach(port => {
                            // We are saying that a port must have a valid
                            // product ID for us to accept it
                            if( port.productId !== undefined ) {
                                portList.push(port.path);
                            }
                        })
                    }
                })
            }
            catch(err) {
                console.log("Failed to retrieve a port list: ", err);
            }

            return portList;
}
```

GazHank, 2021-07-05:
I think we just need to update the example with the recent doc updates at https://serialport.io/docs/guide-installation#invoking-serialport-within-the-renderer-processes

I'd create a PR from my fork but I've added other bells, whistles, tweaks and changes that won't really help anyone; so I'll look to propose a minimal change to the example that just makes it work on the latest versions of electron etc

pg-omm, 2021-09-28:
@GazHank We would like to update our Electron app to the newest version, but `app.allowRenderProcessReuse` [was removed](https://www.electronjs.org/docs/breaking-changes#removed-appallowrendererprocessreuse) on `v14.0.0` of Electron. Are there any plans to make node-serialport [context-aware](https://github.com/electron/electron/issues/18397)?

reconbot, 2021-09-28:
Yes, his draft pr porting us to napi will accomplish this. If you're able to test please do. #2305

aquiot-inc, 2021-11-22:
I struggled with this issue for days! Here is how I "solved" the problem.

I now use Quasar only for an SPA and gave up on electron integration!

For serial port communication, I have a pure Node app that wraps the serial port in an HTTP API with Web Socket for the actual receive/transmit data for the serial port. Then on the SPA client side, I call that API with Axios and client Web Socket. In `quasar dev` mode, the Axios points to my Node dev server. In `quasar build` mode, the node app also serves the html/css/js files. I then deploy this using nexe.

While this isn't a "Quasar problem" per se (Node should have serial port, USB, BLE, etc. all built-in), the end result is that the integration of these various frameworks is very flawed!

I'm sure that there is some version of everything that worked for someone once upon a time. But serial port integration (and all other hardware communications) needs to be part of the release regression testing for the promise of Quasar to be anything more than wishful thinking. Any application framework that cannot handle simple serial port communication out-of-the-box, is a joke! (Just as much a NodeJS/Electron criticism)

I like Quasar as an SPA framework! For anything more than a web app, or something relying on Node functionality that is built-in to Node itself: forget Quasar or any other framework that tries to do the same!

There is also the fact that a simple Quasar/Electron app is well over 100MB! But that is the consequence of the fundamental architecture that I could live with if it just worked. My Node+SPA is well under 1/3 of that size.

Anyway, I hope this will help others, I wish somebody told me this reality before I wasted so much time.

dokodream, 2021-12-14:
I use the electron/Quasar framework and serialport.

I'm downgrading the electron to  v13.0.0.

And referenced[https://stackoverflow.com/questions/62360403/quasar-framework-electron-and-serialport-indexof-error](url)

I ended up packing 55M