serialport/node-serialport #2171: Devices outside of /dev in MacOS

goodoldneon, 2020-11-18:
### Summary of Problem

(Please answer all 3)

**What are you trying to do?**

Create a fake probe using `mknod`, for integration test purposes. MacOS gives `Operation not permitted` when trying to create in `/dev`, so I'd like `node-serialport` to pick up devices outside of `/dev`.

### Versions, Operating System and Hardware

- SerialPort@9.0.2
- Node.js v10.22.1
- Mac
- Hardware and chipset?

reconbot, 2020-11-18:
Like with seriaport list?

romecatholic, 2020-11-18:
I think that is operation problem not serialport‘s

goodoldneon, 2020-11-18:
> Like with seriaport list?

Yea, I'd like to create a virtual device that shows up when I run `npx @serialport/list`. Sorry I'm being so vague. All of this stuff is pretty new to me lol

reconbot, 2020-11-18:
You can make your own binding that wraps the default ones that adds your virtual ports. There are a thousand ways that virtual ports work so I'm not keen on supporting them if they don't show up with the normal serial drivers/system methods.

goodoldneon, 2020-11-19:
Makes sense. Thanks for the responses; I'll close this.

closed 2020-11-19 by goodoldneon