serialport/node-serialport #2172: Can't run the electron app, node version compile error

Lynn-cc, 2020-11-18:
### Summary of Problem

(Please answer all 3)

- What are you trying to do? Run my electron app
- What happens? errors
- What should have happened? run my app

An error occured when I run my electron app. Since last day, it fine. I've tried such as reinstall all the modules reinstall node@10.x , delete package-lock.json and so on , but nothing changed. I see that the serialport/bindings was published a new version yesterday, I don't know if it's the reason, and i reinstall 9.0.1 version is also failed.

### Code to Reproduce the Issue

```js
const serialport = new SerialPort(path)

// Code
```

### Versions, Operating System and Hardware

- SerialPort@?
- Node.js v?
- Windows? Linux? Mac?
- Hardware and chipset? (Prolific/FTDI/Other)

node : v10.23.0
npm: 6.14.8
windows

### error info

App threw an error during load
Error: The module '\\?\D:\workspace\warn-app\node_modules\serialport\node_modules\@serialport\bindings\build\Release\bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 64. This version of Node.js requires
NODE_MODULE_VERSION 85. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.func [as dlopen] (electron/js2c/asar_bundle.js:5:1812)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1203:18)
    at Object.func [as .node] (electron/js2c/asar_bundle.js:5:1812)
    at Module.load (internal/modules/cjs/loader.js:992:32)
    at Module._load (internal/modules/cjs/loader.js:885:14)
    at Function.f._load (electron/js2c/asar_bundle.js:5:12694)
    at Module.require (internal/modules/cjs/loader.js:1032:19)
    at require (internal/modules/cjs/helpers.js:72:18)
    at bindings (D:\workspace\warn-app\node_modules\bindings\bindings.js:112:48)
    at Object.<anonymous> (D:\workspace\warn-app\node_modules\serialport\node_modules\@serialport\bindings\lib\win32.js:1:36)
D:\workspace\warn-app\node_modules\electron\dist\electron.exe exited with signal SIGINT

romecatholic, 2020-11-19:
It seems your 'electron' version and 'abi' version comflict.
check your 'electron' version was:

eg:$./node_modules/.bin/electron -v
v5.0.1

and check your electron abi was:

eg:$./node_modules/.bin/electron -a
70

the abi 70 's nodejs is 12.18.4
reinstall your node and then use command to rebuild serialport by yourself：
npm install -g node-gyp
cd ./node_modules/@serialport/bindings
node-gyp rebuild --target=5.0.1 --arch=x64 --dist-url=https://atom.io/download/electron.

hope these informations can help you！

Lynn-cc, 2020-11-19:
> It seems your 'electron' version and 'abi' version comflict.
> check your 'electron' version was:
>
> eg:$./node_modules/.bin/electron -v
> v5.0.1
>
> and check your electron abi was: [...]

It works ! Thank you very much !!!!

romecatholic, 2020-11-20:
please close this issue。

closed 2020-11-20 by reconbot