serialport/node-serialport #2173: Can't print the list of available ports

zakerl, 2020-11-18:
const SerialPort = require("serialport"); 

let ports = [];
SerialPort.list().then(ports => {
  ports.forEach(function(port) {
    ports.push(port.path);
  }); 
});
console.log(ports);

I've this so far, but my ports is still an empty list, can anyone help?

romecatholic, 2020-11-19:
I think your serialport device driver is not install in your computer

reconbot, 2020-11-19:
Like most promises, they don't resolve right away, they take time to run. So you start the `list()`, and then immediately log the ports array which is empty. `list()` eventually finishes and the fills the array.

This will work
```js
SerialPort.list().then(ports => {
  console.log(ports)
})
```

```js
async function listPorts() {
  const ports = await SerialPort.list()
  console.log(ports)
}

listPorts()
```

Will also work.

closed 2020-11-19 by reconbot