serialport/node-serialport #2176: can't install

JamesBlakeyMilner, 2020-11-22:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?
Install on WIndows 10
- What happens?
Install fails
- What should have happened?
Install correctly
### Code to Reproduce the Issue

npm i serialport

PS C:\fucksakes> npm init
This utility will walk you through creating a package.json file.
It only covers the most common items, and tries to guess sensible defaults.

See `npm help init` for definitive documentation on these fields
and exactly what they do.

Use `npm install <pkg>` afterwards to install a package and
save it as a dependency in the package.json file.

Press ^C at any time to quit.
package name: (fucksakes)
version: (1.0.0)
description:
entry point: (index.js)
test command:
git repository:
keywords:
author:
license: (ISC)
About to write to C:\fucksakes\package.json:

{
  "name": "fucksakes",
  "version": "1.0.0",
  "description": "",
  "main": "index.js",
  "scripts": {
    "test": "echo \"Error: no test specified\" && exit 1"
  },
  "author": "",
  "license": "ISC"
}

Is this OK? (yes)
PS C:\fucksakes> npm i serialport

> @serialport/bindings@9.0.2 install C:\fucksakes\node_modules\@serialport\bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=14.15.1 runtime=node arch=x64 libc= platform=win32)

C:\fucksakes\node_modules\@serialport\bindings>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild )
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  serialport.cpp
  serialport_win.cpp
  win_delay_load_hook.cc
C:\\Users\\james\\AppData\\Local\\node-gyp\\Cache\\14.15.1\\x64\\node.lib : fatal error LNK1106: invalid file or disk f
ull: cannot seek to 0x376F20 [C:\fucksakes\node_modules\@serialport\bindings\build\bindings.vcxproj]
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\MSBuild\15.0\Bin\MSBuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)
gyp ERR! System Windows_NT 10.0.18363
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\fucksakes\node_modules\@serialport\bindings
gyp ERR! node -v v14.15.1
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok
npm WARN fucksakes@1.0.0 No description
npm WARN fucksakes@1.0.0 No repository field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@9.0.2 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the @serialport/bindings@9.0.2 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\james\AppData\Roaming\npm-cache\_logs\2020-11-22T14_34_59_864Z-debug.log

### Versions, Operating System and Hardware

- SerialPort@?
- Node.js v?14.15.1
- Windows? Linux? Mac? Windows 10
- Hardware and chipset? (Prolific/FTDI/Other)

reconbot, 2020-12-04:
try `serialport@9.0.3`

bertaveira, 2020-12-29:
I had the same problem but installing the 9.0.3 version solved it for me just like @reconbot said

GazHank, 2021-07-05:
I believe this relatess to historic issues with old releases, is resolved in recent version.  Please shout if you are still facing issues

reconbot, 2022-04-06:
Serialport@10 leverages napi and distributes most platforms in the package so now download. It generally has better support.

I can't really tell what your issues is @nflachaire it looks like you're having issues with installing v10, but we need more information and I'd ask you open a new issue and fill out the form.

closed 2022-04-06 by reconbot