serialport/node-serialport #2179: cannot access the data on health card reader

powlsn, 2020-11-26:
### Summary of Problem

I'm trying to access the data of a health card with this device [card reader](https://www.ingenico.de/healthcare/downloads/archiv/orga6041.html) which is connected via USB. This device comes with its own drivers for Windows, which I have already installed.
When I use `Serialport.list()` to display the available devices, I see that it is recognized. I can open the port on (COM3) in my case, but when I insert a card into the device for reading, no data is received.
If I have used the api correctly, I expect at least some incomming data

### Code to Reproduce the Issue

```js
const port = new SerialPort(path)

port.on('open', err => {
  if (err) return console.log("Error on Open: " + err.message);
  console.log("COM3 Open");

  port.on('data', data => {
    console.log('data: ' + data.toString());
  });
});
```

i also tryed

```js
const port = new SerialPort('COM3');
const parser = port.pipe(new Ready({delimiter: 'READY'})).pipe(new Readline({delimiter: '\r\n'}));

port.on('open', err => {
  console.log('Open');
  if (err) console.log(err);
});
parser.on('ready', () => console.log('ready byte sequence has been received'));
parser.on('data', console.log);
parser.on('error', err => { console.log(err)});
```

### Versions, Operating System and Hardware

- SerialPort@ 9.0.2
- Node.js v14.8.0 
- Windows 10 build19042.572
- Ingenico - ORGA 6041 with official win driver, connected via USB

i am grateful for any advice on how to use this amazing package. please keep in mind i am a junior developer and not very familiar with

sjlongland, 2020-12-12:
My German is non-existant, so the page you linked to is useless to me, but maybe the device is expecting you to send it some commands first before it'll just spit out data?

You'll need to contact the company to find out what those commands are.  Then, it should be a case of doing some `port.write()` calls with appropriate `Buffer` objects containing the commands needed to "wake it up".

GazHank, 2021-07-05:
Hi @powlsn 

did you have progress with this issue, or identifying the command messages that you need to send to the card reader for it to talk to you?

powlsn, 2021-07-18:
Hi @GazHank so far i've not solved it.
I tried to contact the company as @sjlongland suggested. But they only refer to the official documentation of 'gematik' and this is not very helpful. The device is a 'mobile' card reader specified by 'gematik' and the connection examples are implemented with CT-API. The device supports PCSC and i also get a status word 67 00 back, but not much else. I get the device name and the card slot and the event for the respective card slot is also thrown, but I get no data from the device except sw 6700 which according to the specification stands for 'command not allowed'.