serialport/node-serialport #2184: electron  electron-builder  http 404 https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@9.0.2/bindings-v9.0.2-electron-v82-win32-x64.tar.gz

syanfox, 2020-12-03:
electron version:
10.1.5

node version:
12.19.0

serialport version:
9.0.2

error info:
 prebuild-install http request GET https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@9.0.2/bindings-v9.0.2-electron-v82-win32-x64.tar.gz
    prebuild-install http 404 https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@9.0.2/bindings-v9.0.2-electron-v82-win32-x64.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=10.1.5 runtime=electron arch=x64 libc= platform=win32)

reconbot, 2020-12-04:
try `serialport@9.0.3`

robincsamuel, 2021-05-26:
@reconbot I'm getting the same issue but a different version. 

I've installed `serialport@9.0.3` as you suggested, however, it's still trying to pull @serialport/bindings@9.0.8`.

Electron version: 
10.4.7

Node version:
14.17.0

serial port version:
9.0.3 
(I've tried with 9.0.7 as well, it works fine on `8.0.7`.

Logs:

```
⨯ cannot build native dependency  reason=prebuild-install failed with error and build from sources not possible because platform or arch not compatible
                                    cause=exit status 1
                                    errorOut=prebuild-install info begin Prebuild-install version 6.1.2
    prebuild-install WARN install prebuilt binaries enforced with --force!
    prebuild-install WARN install prebuilt binaries may be out of date!
    prebuild-install info looking for local prebuild @ prebuilds/bindings-v9.0.8-electron-v82-win32-x64.tar.gz
    prebuild-install info looking for cached prebuild @ /Users/robin/.npm/_prebuilds/876058-bindings-v9.0.8-electron-v82-win32-x64.tar.gz
    prebuild-install http request GET https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@9.0.8/bindings-v9.0.8-electron-v82-win32-x64.tar.gz
    prebuild-install http 404 https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@9.0.8/bindings-v9.0.8-electron-v82-win32-x64.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=10.4.7 runtime=electron arch=x64 libc= platform=win32)
    
                                    command=/usr/local/Cellar/node/16.1.0/bin/node /Users/robin/Projects/api-scan/node_modules/prebuild-install/bin.js --platform=win32 --arch=x64 --target=10.4.7 --runtime=electron --verbose --force
                                    workingDir=/Users/robin/Projects/api-scan/node_modules/@serialport/bindings
  • Above command failed, retrying 5 more times
```

robincsamuel, 2021-05-26:
Okay, I realized the `serialport@9.03` has dependency  `@serialport/bindings": "^9.0.4` which finds the latest version `9.0.8`. And that doesn't have prebuilt binaries. 

Any idea?

robincsamuel, 2021-05-26:
Sorry to spam, but I think this will help someone. 

Though I was able to get rid of the build time errors with the above workaround, there were runtime errors and I feel it's not reliable since I don't know the reason.

Further research leads me to yarn's [selective dependency resolution](https://classic.yarnpkg.com/en/docs/selective-version-resolutions/) and that worked.

Added this to the package.json and everything works as expected.
```
  "resolutions": {
    "@serialport/bindings": "9.0.7"
  },
```

reconbot, 2021-05-26:
I just fixed the binaries for 9.0.8 [they're building now](https://github.com/serialport/node-serialport/actions/runs/878496051) which might lead to it working for you, I'd try to use the same serialport version as bindings. Technically it should work as we follow semver but they're only tested in lockstep.

robincsamuel, 2021-05-26:
Awesome!

Locking it to 9.0.7 works for me. But yeah, it’d great if the same versions can be used.

Thanks much :)

reconbot, 2021-05-26:
9.0.8 is now published btw

closed 2021-05-26 by reconbot