serialport/node-serialport #2185: serialport-list  结果异常，怎么解决

ydduan123, 2020-12-04:
[image]
系统版本
root@localhost:~# cat /etc/issue
Ubuntu 16.04.6 LTS \n \l

遇到的问题
root@localhost:~# serialport-list -f json
{"errno":-2,"code":"ENOENT","syscall":"spawn udevadm","path":"udevadm","spawnargs":["info","-e"]}

reconbot, 2020-12-04:
`udevadm` needs to be installed

ArthurMbraga, 2020-12-10:
I'm having the same issue, how can I install that on Android/linux?

reconbot, 2020-12-17:
I don't know, but I do think it's part of the `udev` subsytem on most linux;s

ArthurMbraga, 2020-12-29:
I tried to solve it, but it was an Android/linux permission problem. 
For someone who is in the same situation, I haven't tested it, but my solution would be to root the phone.

closed 2022-04-28 by stale[bot]