serialport/node-serialport #2187: Missing precompiled binaries on Windows

pedro-w, 2020-12-09:
### Summary of Problem
I am installing an NPM package with serialport as a dependency. Relevant part of package.json
```
  "dependencies": {
    "core-js": "^3.8.1",
    "express": "^4.17.1",
    "random": "^2.2.0",
    "regenerator-runtime": "^0.13.7",
    "serialport": "^9.0.3",
    "sprintf-js": "^1.1.2",
    "uplot": "^1.4.7"
  },
```
I get
```
> @serialport/bindings@9.0.3 install D:\temp\ser\node_modules\@serialport\bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=14.15.0 runtime=node arch=x64 libc= platform=win32)
```

As far as I can tell, node 14.x (ABI 83) should be supported (current node LTS version is 14.15.1). Ideally, I would prefer not to install C compilers to compile my own binaries.

I am not an expert on this, please advise.

### Versions, Operating System and Hardware

- SerialPort@9.0.3
- Node.js v14.15.0
- Windows 10 Pro 20H2 19042.685

reconbot, 2020-12-09:
We currently don’t have a working way to build for win32 only win64.

pedro-w, 2020-12-09:
Thanks for the swift reply. Just to confirm, this is because my node is 32-bit (even though my Windows is 64 bit), so I should update to 64-bit node if I can?

pedro-w, 2020-12-09:
All right, using the 64-bit node seems to work.

closed 2020-12-09 by pedro-w