serialport/node-serialport #2189: How to send password required for login

Vn-ChemGio, 2020-12-14:
Hey guy!
I'm have being creat nodeJs app for monitor and controll some old device
But I have a problem
When using Hyper Terminal to connect device, after call connect, My device will require a Password, and I have to input device's password to verify (Password store in device)
So, How can I send /write .... password to device when create a instance SerialPort or open connect ....
Thank for help !

Here is my example use case in Hyper terminal

[image]

GazHank, 2021-07-05:
Hi @Vn-ChemGio are you still facing issues with this?  I'd assume you just need to open the port and call a read or write operations, but perhaps I'm missing something

Vn-ChemGio, 2021-08-25:
I were fixing this issue !
by send "enter" symbol to device after connected
Because this device always stored last my input after disconected

closed 2022-04-28 by stale[bot]