serialport/node-serialport #2191: TypeError: Third argument must be a function

Lovinity, 2020-12-19:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?: Open a connection to a serial port device via the serialport constructor on Windows 10, electron 11.1, bindings 9.0.4.
- What happens? When attempting to establish a connection to a serialport, it throws an error saying third argument must be a function.
- What should have happened? The connection establishes successfully and we are able to begin writing (which was the case in electron 9 and bindings 9.0.2).

### Code to Reproduce the Issue

```js

// let device = device path retrieved from electron.settings, which I verified is operating correctly.

delaySerial = new serialport(device, {
	baudRate: 38400,
});

delaySerial.on("error", (err) => {
	serialError(err); // function sends the error both to the console and another logging utility. It also disconnects the serialport if not already disconnected and attempts to reconnect after a 15 second delay.
});
```

### Versions, Operating System and Hardware

- SerialPort@? 9.0.4
- Node.js v? Electron 11.1.0
- Windows? Linux? Mac? Windows 10 build 2004
- Hardware and chipset? (Prolific/FTDI/Other) Unknown/other (uses USB connection, but this error occurs no matter which COM port I try)

orange-beans, 2020-12-19:
@Lovinity Encountered same problem. This problem was not found in serialport@9.0.3 + electron@8 though.

pklaschka, 2020-12-19:
Just for reference (this isn't really a solution, of course, if anything, it's an ugly hack): After encountering the same error, I was able to get it to work with Electron 11.1.0 and serialport in version 8.0.8 (after running `electron-rebuild`, of course) on both Windows and Linux.

Seeing

> @Lovinity Encountered same problem. This problem was not found in serialport@9.0.3 + electron@8 though.

might, of course, mean, that a downgrade to an exact 9.0.3 might also have solved that problem, of course (I probably didn't properly specify the exact version, which has then lead me to believe 9.0.3 wasn't working either 😅 ).

Lovinity, 2020-12-19:
@pklaschka That's true it's a dirty solution. I unfortunately cannot rely on rebuild and must use prebuilt binaries due to limitations with concurrent jobs in the build tool I use (if I use matrices, not only does that drastically complicate things, but it might cost money too; so I have to build for all OSes on MacOS. And rebuild will only work for the MacOS build on MacOS).

reconbot, 2020-12-20:
@pklaschka so you're saying the prebuilds for windows in v9.0.4 have an issue but if you rebuild them they work for you?

pklaschka, 2020-12-21:
> @pklaschka so you're saying the prebuilds for windows in v9.0.4 have an issue but if you rebuild them they work for you?

No. With 9.0.4, the "Third argument must be a function" error occurs either way (also after rebuilding). I just mentioned `electron-rebuild` to make it clear that, to use Electron 11, I had rebuilt the older (!) version of `serialport` with success. Sorry if that wasn't clear 😉

reconbot, 2020-12-21:
That's frustrating, nothing about the serialport code has changed between those versions.

Lovinity, 2020-12-21:
Maybe a dependency update is breaking the build?

reconbot, 2020-12-21:
Yeah, probably Nan which was required to build for the latest electron

https://github.com/serialport/node-serialport/blob/master/packages/bindings/src/serialport_win.cpp#L314-L317 is probably the code you're hitting. It's the write callback, which is guaranteed to have a callback here https://github.com/serialport/node-serialport/blob/master/packages/bindings/lib/win32.js#L10

I'll take a peak at some of the build warnings from the c++, maybe they've changed how functions work or something.

naturalfreak, 2020-12-21:
Same problem

gbmhunter, 2020-12-23:
Same problem here. It used to work fine, and then I started playing around with `electron-rebuild` and called it a number of times. Now I get this exact same error. Sorry this is vague, I can't really give any more details about it.

Kallontz, 2020-12-24:
I encountered same problem, too and I found the reason that cause this issue.

I used Electron 11.1, @serialport/bindings 9.0.4, electron-builder 22.9.1.

The reason what occurs this issue is wrong prebuilt image of @serialport/bindings@9.0.4.

By default, `electron-builder` using prebuilt image if exists. And there was a prebuilt image of @serialport/bindings.

When I run the application after `electron-rebuild` was **successfully** working serialport@9.0.4 with @serialport/bindings@9.0.4.

But after run the `electron-builder`, this issue **reproduced** with no code changes.

And I looked into the logs of `electron-builder`, it uses prebuilt image of @serialport/bindings.

So I add the option to force rebuild dependencies to the `electron-builder` option shots the problem.

```
"build": {
  ...
  "buildDependenciesFromSource": true,
  ...
}
```

Packing time was slightly slowed-down but it was my best solution to using the latest version of serialport.

reconbot, 2020-12-24:
Well... that's promising, it means a problem with the build vs the source.  I removed the existing builds and [forced another one](https://github.com/serialport/node-serialport/runs/1606635637?check_suite_focus=true) can you try it again?

orange-beans, 2020-12-25:
> Well... that's promising, it means a problem with the build vs the source. I removed the existing builds and [forced another one](https://github.com/serialport/node-serialport/runs/1606635637?check_suite_focus=true) can you try it again?

Tried with the latest built binding (bindings-v9.0.4-electron-v85-win32-x64), the "TypeError: Third argument must be a function" error still occurs.

Tried to run electron-rebuild for serialport module, also failed #2194

liyatang, 2020-12-25:
um， work for me change electron version to 10.1.3。

Kallontz, 2020-12-28:
> Well... that's promising, it means a problem with the build vs the source. I removed the existing builds and [forced another one](https://github.com/serialport/node-serialport/runs/1606635637?check_suite_focus=true) can you try it again?

I just got a same result as @orange-beans after flush the npm cache.

```
  • install prebuilt binary  name=@serialport/bindings version=9.0.4 platform=win32 arch=x64
  • execute command  command='{nodejs}\node.exe' '{Project}\node_modules\prebuild-install\bin.js' --platform=win32 --arch=x64 --target=11.1.1 --runtime=electron --verbose --force
                     workingDirectory={Project}\node_modules\@serialport\bindings
  • command executed  executable={nodejs}\node.exe
                      errorOut=prebuild-install info begin Prebuild-install version 6.0.0
    prebuild-install WARN install prebuilt binaries enforced with --force!
    prebuild-install WARN install prebuilt binaries may be out of date!
    prebuild-install info looking for cached prebuild @ {AppData\Roaming}\npm-cache\_prebuilds\a203aa-bindings-v9.0.4-electron-v85-win32-x64.tar.gz
    prebuild-install http request GET https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@9.0.4/bindings-v9.0.4-electron-v85-win32-x64.tar.gz
    prebuild-install http 200 https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@9.0.4/bindings-v9.0.4-electron-v85-win32-x64.tar.gz
    prebuild-install info downloading to @ {AppData\Roaming}\npm-cache\_prebuilds\a203aa-bindings-v9.0.4-electron-v85-win32-x64.tar.gz.35396-0b268dadf342e.tmp
    prebuild-install info renaming to @ {AppData\Roaming}\npm-cache\_prebuilds\a203aa-bindings-v9.0.4-electron-v85-win32-x64.tar.gz
    prebuild-install info unpacking @ {AppData\Roaming}\npm-cache\_prebuilds\a203aa-bindings-v9.0.4-electron-v85-win32-x64.tar.gz
    prebuild-install info unpack resolved to {Project}\node_modules\@serialport\bindings\build\Release\bindings.node
    prebuild-install info install Successfully installed prebuilt binary!
```

Lovinity, 2021-01-10:
I think the problem is you need to bump the version up. Those of us who already downloaded 9.0.4 have a cache of it, and npm is not going to fetch the corrected build binary unless the version is bumped to 9.0.5.

Erik7354, 2021-01-11:
Encountered the same problem with Electron 11.1.1 and Serialport 9.0.4.

Adding `"rebuild":  "electron-rebuild -f -w serialport"` to scripts and running `npm run rebuild` solved it for me.

TonyStark106, 2021-02-22:
Add
```
"build": {
  ...
  "npmRebuild": false,
  ...
}
```
And run `electron-rebuild -f -w serialport` work for me ([two-package-structure](https://www.electron.build/tutorials/two-package-structure) should run `electron-rebuild -m app -f -w serialport` too). But cross compiling seem broken. Maybe a temporary solution for Electron 11.

sammlapp, 2021-02-25:
I am encountering this issue as well. I have Electron 11.2.3 and Serialport 9.0.4. 
Are any of these solutions "recommended" or consistently working? I have tried different serialport and electron versions without success. @TonyStark106  is the idea in your solution to add this "build" option to to package.json?

TonyStark106, 2021-02-25:
Yes! Like this:
```
{
  "scripts": {
    "postinstall": "electron-builder install-app-deps",
    "rebuild": "electron-rebuild -f -w serialport"
  },
  "dependencies": {
    "serialport": "9.0.4"
  },
  "devDependencies": {
    "electron": "11.2.3",
    "electron-builder": "22.9.1",
    "electron-rebuild": "2.3.5"
  },
  "build": {
    "npmRebuild": false
  }
}
```
Run `yarn install && yarn rebuild` will fix `TypeError: Third argument must be a function` when developing. Just Like @Erik7354 say. But i encountered this issue again when distributing a release bundle unless add "npmRebuild" option simultaneously.

jgstephe, 2021-04-07:
FWIW, upgrading from Electron 11 to Electron 12.0.2 solved this issue for me.

Julusian, 2021-04-14:
I have been digging and have an interesting discovery here.

I have created a very minimal project, which simply tries to open a serialport. And I can reproduce the issue when run with `ELECTRON_RUN_AS_NODE=1 npx electron@^11.0.0 index.js`

If inside `node_modules/@serialport/bindings` I issue:
1. `yarn add prebuild` (its not listed as a dependency)
1. `yarn node-gyp rebuild --target=11.0.0-beta.12 --target_arch=x64 --runtime=electron --dist-url=https://ato
m.io/download/electron` to force a local build

Then back in the the root folder I try the `ELECTRON_RUN_AS_NODE=1 npx electron@^11.0.0 index.js` again, it works! It is also worth noting that a  `ELECTRON_RUN_AS_NODE=1 npx electron@11.0.0-beta.11 index.js` does not work

But, if I try it with the rebuild being `--target=11.0.0-beta.11`, then it breaks again. But `ELECTRON_RUN_AS_NODE=1 npx electron@11.0.0-beta.11 index.js` now works.

At this point I was starting to be suspicious that beta.12 introduced a breaking change in the abi. For reference the node-abi project uses `11.0.0-beta.11` as the version which stabilised the abi, and so that is the version that prebuild used when building the binaries for this project.

Some skimming of the git diff between those electron versions later, and I found this change https://github.com/electron/electron/compare/v11.0.0-beta.11...v11.0.0-beta.12#diff-5115fa205eb6e02fcf3815ec5098c8bb6e1de0f29b5c23425349f9db1b6ca6ec
So I manually applied that patch to my node-gyp cache of the headers (in C:\Users\Julus\AppData\Local\node-gyp\Cache\11.0.0-beta.11\include\node\common.gypi) and then repeat the rebuild from above.
Try running it and it works again!

So it looks like node-abi is giving out a version number that is broken. `electron-rebuild` will be working because it will build against the exact version of electron you are using.
Also https://github.com/cmake-js/cmake-js/issues/222#issuecomment-732746074 appears to have discovered the same missing define.

As for solutions, the quickest option would be to add `'defines': ['V8_REVERSE_JSARGS'],` to binding.gyp. Probably behind a check to only apply to electron 11, to minimise risk of other breakages. I will look into a PR for this tomorrow.
It would be a good idea to chase this upstream in node-abi, prebuild or electron, but it would be good to have some other native libraries that are broken in the same way to help prove where the issue lies. Does anyone know of any other nan based libraries using prebuild with prebuilds made for electron 11?

samjonesigd, 2021-04-14:
Hi all

When i do a rebuild it's fine on electron:serve but when I go to build it as a windows or mac app I get the error in the app.

Is there anything I can do to get around this for now?

mussacharles60, 2021-05-04:
This one `velectron-rebuild -f -w serialport` works for me on `electron@v11.2.3` and `serialport@9.0.7`

JeongJun-Lee, 2021-05-22:
same here.  how can i always force rebuild this in prod build script whenever production build?

closed 2021-05-24 by reconbot

samjonesigd, 2021-06-01:
For anyone reading I eventually got this working by using electron 10 and serialport 9.0.3 & rebuilding with electron-rebuild

chenzeshu, 2021-06-29:
win10
 "dependencies": {
     "serialport": "^9.0.7",
  },
  "devDependencies": {
    "electron": "11.0.0",
  },

**"electron-rebuild -f -w serialport"   works for me!**
thx!!

PR #2237 (2021-05-24): fix: electron 11 prebuilds broken
Fixes #2191 

A temporary fix until fixed upstream https://github.com/lgeiger/node-abi/issues/103 or https://github.com/prebuild/prebuild/issues/275

As explained in my comment https://github.com/serialport/node-serialport/issues/2191#issuecomment-819149005, electron v11.0.0-beta.12 started to define the variable `V8_REVERSE_JSARGS`. But as prebuild/node-abi use v11.0.0-beta.11 for producing the prebuilds, this resulted in binaries which were broken.

This is the simplest way I could find to do this while keeping it scoped to just electron 11. I couldn't find  [... cut]