serialport/node-serialport #2193: is not a valid win32 Application

Edig, 2020-12-23:
I have installed node-serialport since V8 a few days ago I update it to V9.0.2 with Electron V9.4.0.

But I update serialport  again and it change to V9.0.4 I roll-back to V9.0.2 and since then I have this problem when I run the app in windows, even if I try `electron ./` directly in windows. 

`bindings\build\Release\bindings.node is not a valid Win32 application.`

If I run a build in windows that was builded it in Mac, the app just crash without any error when I require Serial port. I try it in a new app and I have the same problem, I've try many electron versions different and have the same problem

Note: If I run this on a Mac I works perfectly, just can't make it work in Windows

here is my example code. 

main
`const { app, BrowserWindow } = require('electron')

function createWindow () {

    app.allowRendererProcessReuse = false

  const win = new BrowserWindow({
    width: 800,
    height: 600,
    webPreferences: {
      nodeIntegration: true
    }
  })

  win.loadFile('index.html')
}

app.whenReady().then(createWindow)

app.on('window-all-closed', () => {
  if (process.platform !== 'darwin') {
    app.quit()
  }
})

app.on('activate', () => {
  if (BrowserWindow.getAllWindows().length === 0) {
    createWindow()
  }
})`
I

index
`<script>

    alert('a')

    const SerialPort = require('serialport')
    const Readline = require('@serialport/parser-readline')
    const Ready = require('@serialport/parser-ready') 
    alert('b')
</script>

Hello World`

GazHank, 2021-07-05:
Hi @Edig are you still facing issues on the latest versions?

Edig, 2021-07-05:
Hi, I instal electron 9.x again  and re-install all the plugins, and rebuild everything. And it works

GazHank, 2021-07-05:
Thanks great @Edig thanks for the quick response :-)

closed 2021-07-05 by GazHank