serialport/node-serialport #2194: electron-rebuild failed with syntax errors

orange-beans, 2020-12-25:
### Summary of Problem

Running electron-rebuild for serialport module failed. And lots of syntax errors appears in the error message.

### Code to Reproduce the Issue

```js
$ yarn electron-rebuild
yarn run v1.22.4
$ node -r ../internals/scripts/BabelRegister.js ../internals/scripts/ElectronRebuild.js
× Rebuild Failed

An unhandled error occurred inside electron-rebuild
gyp info it worked if it ends with ok
gyp info using node-gyp@6.1.0
gyp info using node@12.18.4 | win32 | x64
gyp info find Python using Python version 2.7.17 found at "C:\Python27\python.exe"
gyp info find VS using VS2015 (14.0) found at:
gyp info find VS "C:\Program Files (x86)\Microsoft Visual Studio 14.0"
gyp info find VS run with --verbose for detailed information
gyp info spawn C:\Python27\python.exe
gyp info spawn args [
gyp info spawn args   'D:\\Projects\\Desktop-projects\\Electron-Boilerplates\\electron-boilplate-v140\\node_modul
es\\electron-rebuild\\node_modules\\node-gyp\\gyp\\gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'msvs',
gyp info spawn args   '-I',
gyp info spawn args   'D:\\Projects\\Desktop-projects\\Electron-Boilerplates\\electron-boilplate-v140\\app\\node_
modules\\@serialport\\bindings\\build\\config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'D:\\Projects\\Desktop-projects\\Electron-Boilerplates\\electron-boilplate-v140\\node_modul
es\\electron-rebuild\\node_modules\\node-gyp\\addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\chengcheng\\.electron-gyp\\11.0.3\\include\\node\\common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=C:\\Users\\chengcheng\\.electron-gyp\\11.0.3',
gyp info spawn args   '-Dnode_gyp_dir=D:\\Projects\\Desktop-projects\\Electron-Boilerplates\\electron-boilplate-v
140\\node_modules\\electron-rebuild\\node_modules\\node-gyp',
gyp info spawn args   '-Dnode_lib_file=C:\\\\Users\\\\chengcheng\\\\.electron-gyp\\\\11.0.3\\\\<(target_arch)\\\\
node.lib',
gyp info spawn args   '-Dmodule_root_dir=D:\\Projects\\Desktop-projects\\Electron-Boilerplates\\electron-boilplat
e-v140\\app\\node_modules\\@serialport\\bindings',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'D:\\Projects\\Desktop-projects\\Electron-Boilerplates\\electron-boilplate-v140\\app\\node_
modules\\@serialport\\bindings\\build',
gyp info spawn args   '-Goutput_dir=.'
gyp info spawn args ]
gyp info spawn C:\Program Files (x86)\MSBuild\14.0\bin\MSBuild.exe
gyp info spawn args [
gyp info spawn args   'build/binding.sln',
gyp info spawn args   '/clp:Verbosity=minimal',
gyp info spawn args   '/nologo',
gyp info spawn args   '/p:Configuration=Release;Platform=x64'
gyp info spawn args ]
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  serialport.cpp
  serialport_win.cpp
  win_delay_load_hook.cc
c:\users\chengcheng\.electron-gyp\11.0.3\include\node\cppgc/common.h(17): error C2144: syntax error: 'int' should
 be preceded by '}' (compiling source file ..\src\serialport_win.cpp) [D:\Projects\Desktop-projects\Electron-Boil
erplates\electron-boilplate-v140\app\node_modules\@serialport\bindings\build\bindings.vcxproj]
c:\users\chengcheng\.electron-gyp\11.0.3\include\node\cppgc/common.h(17): error C2144: syntax error: 'int' should
 be preceded by '}' (compiling source file ..\src\serialport.cpp) [D:\Projects\Desktop-projects\Electron-Boilerpl
ates\electron-boilplate-v140\app\node_modules\@serialport\bindings\build\bindings.vcxproj]
c:\users\chengcheng\.electron-gyp\11.0.3\include\node\cppgc/common.h(17): error C2144: syntax error: 'int' should
[... 48 lines cut ...]
efore '{' (compiling source file ..\src\serialport.cpp) [D:\Projects\Desktop-projects\Electron-Boilerplates\elect
ron-boilplate-v140\app\node_modules\@serialport\bindings\build\bindings.vcxproj]
c:\users\chengcheng\.electron-gyp\11.0.3\include\node\v8-internal.h(16): error C2143: syntax error: missing ';' b
efore '{' (compiling source file ..\src\serialport_win.cpp) [D:\Projects\Desktop-projects\Electron-Boilerplates\e
lectron-boilplate-v140\app\node_modules\@serialport\bindings\build\bindings.vcxproj]
c:\users\chengcheng\.electron-gyp\11.0.3\include\node\v8-internal.h(16): error C2447: '{': missing function heade
r (old-style formal list?) (compiling source file ..\src\serialport.cpp) [D:\Projects\Desktop-projects\Electron-B
oilerplates\electron-boilplate-v140\app\node_modules\@serialport\bindings\build\bindings.vcxproj]
c:\users\chengcheng\.electron-gyp\11.0.3\include\node\v8-internal.h(16): error C2447: '{': missing function heade
r (old-style formal list?) (compiling source file ..\src\serialport_win.cpp) [D:\Projects\Desktop-projects\Electr
on-Boilerplates\electron-boilplate-v140\app\node_modules\@serialport\bindings\build\bindings.vcxproj]
c:\users\chengcheng\.electron-gyp\11.0.3\include\node\v8.h(1224): error C2039: 'Address': is not a member of 'v8:
:internal' (compiling source file ..\src\serialport.cpp)c:\users\chengcheng\.electron-gyp\11.0.3\include\node\v8.
h(1224): error C2039: 'Address': is not a member of 'v8::internal' (compiling source file ..\src\serialport_win.c
pp) [D:\Projects\Desktop-projects\Electron-Boilerplates\electron-boilplate-v140\app\node_modules\@serialport\bind
ings\build\bindings.vcxproj]

```

### Versions, Operating System and Hardware

- SerialPort v9.0.4
- Electron v11.0.3
- Node.js v12.18.1
- Windows 10 64bit

hustzju, 2021-01-31:
I have same problem when compile serialport@9.0.4 with electron V11 .When I downgrade electron to V9, it works,
It seems that serialport doesnot support electron version >V10

GazHank, 2021-07-05:
Based on the error messages, I think this may be use to the use of Python 2.  The docs have since been updated to correctly reflect the use of python 3.

This should now work with Electron 11-13; but if you are still facing any issues then please let me know

gunnarSensus, 2021-07-20:
@GazHank im getting this same error in an electron project of mine, but i am fairly sure i have gyp pointing to the correct python (3)

i have my .npmrc set with the correct python:
```
msvs_version=2019
python=C:\Users\rannug\AppData\Local\Microsoft\WindowsApps\python3.ex
```

and i also tried setting the PYTHON environment variable to my python3 exe, but no luck. here is the output of my electron-rebuild (npm install completed successfully without error)
```
PS C:\Users\rannug\Documents\work\local-configurator> $env:PYTHON = 'C:\Users\rannug\AppData\Local\Microsoft\WindowsApps\python3.exe'
PS C:\Users\rannug\Documents\work\local-configurator> .\node_modules\.bin\electron-rebuild
⠸ Building module: bindings, Completed: 0gyp info find Python using Python version 3.9.6 found at "C:\Users\rannug\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\python.exe"
⠏ Building module: bindings, Completed: 0gyp info find VS using VS2015 (14.0) found at:
gyp info find VS "C:\Program Files (x86)\Microsoft Visual Studio 14.0"
gyp info find VS run with --verbose for detailed information
gyp info spawn C:\Users\rannug\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\python.exe
gyp info spawn args [
gyp info spawn args   'C:\\Users\\rannug\\Documents\\work\\local-configurator\\node_modules\\electron-rebuild\\node_modules\\node-gyp\\gyp\\gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'msvs',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\rannug\\Documents\\work\\local-configurator\\node_modules\\@serialport\\bindings\\build\\config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\rannug\\Documents\\work\\local-configurator\\node_modules\\electron-rebuild\\node_modules\\node-gyp\\addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\rannug\\.electron-gyp\\13.1.7\\include\\node\\common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=C:\\Users\\rannug\\.electron-gyp\\13.1.7',
gyp info spawn args   '-Dnode_gyp_dir=C:\\Users\\rannug\\Documents\\work\\local-configurator\\node_modules\\electron-rebuild\\node_modules\\node-gyp',
gyp info spawn args   '-Dnode_lib_file=C:\\\\Users\\\\rannug\\\\.electron-gyp\\\\13.1.7\\\\<(target_arch)\\\\node.lib',
gyp info spawn args   '-Dmodule_root_dir=C:\\Users\\rannug\\Documents\\work\\local-configurator\\node_modules\\@serialport\\bindings',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'C:\\Users\\rannug\\Documents\\work\\local-configurator\\node_modules\\@serialport\\bindings\\build',
gyp info spawn args   '-Goutput_dir=.'
gyp info spawn args ]
⠼ Building module: bindings, Completed: 0gyp info spawn C:\Program Files (x86)\MSBuild\14.0\bin\MSBuild.exe
gyp info spawn args [
gyp info spawn args   'build/binding.sln',
gyp info spawn args   '/clp:Verbosity=minimal',
gyp info spawn args   '/nologo',
gyp info spawn args   '/p:Configuration=Release;Platform=x64',
gyp info spawn args   map: [Function (anonymous)]
gyp info spawn args ]
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
⠇ Building module: bindings, Completed: 0  serialport.cpp
⠸ Building module: bindings, Completed: 0c:\users\rannug\.electron-gyp\13.1.7\include\node\v8.h(1652): error C2059: syntax error: '' [C:\Users\rannug\Documents\
work\local-configurator\node_modules\@serialport\bindings\build\bindings.vcxproj]
c:\users\rannug\.electron-gyp\13.1.7\include\node\v8.h(1654): error C2238: unexpected token(s) preceding ';' [C:\Users\
rannug\Documents\work\local-configurator\node_modules\@serialport\bindings\build\bindings.vcxproj]
c:\users\rannug\.electron-gyp\13.1.7\include\node\v8.h(1670): error C2061: syntax error: identifier 'ResolveCallback' [
C:\Users\rannug\Documents\work\local-configurator\node_modules\@serialport\bindings\build\bindings.vcxproj]
c:\users\rannug\.electron-gyp\13.1.7\include\node\v8.h(7292): error C2059: syntax error: '' [C:\Users\rannug\Documents\
work\local-configurator\node_modules\@serialport\bindings\build\bindings.vcxproj]
c:\users\rannug\.electron-gyp\13.1.7\include\node\v8.h(8652): error C2061: syntax error: identifier 'HostImportModuleDy
namicallyCallback' [C:\Users\rannug\Documents\work\local-configurator\node_modules\@serialport\bindings\build\bindings.
vcxproj]
✖ Rebuild Failed

An unhandled error occurred inside electron-rebuild
node-gyp failed to rebuild 'C:\Users\rannug\Documents\work\local-configurator\node_modules\@serialport\bindings'.
Error: `C:\Program Files (x86)\MSBuild\14.0\bin\MSBuild.exe` failed with exit code: 1

Error: node-gyp failed to rebuild 'C:\Users\rannug\Documents\work\local-configurator\node_modules\@serialport\bindings'.
Error: `C:\Program Files (x86)\MSBuild\14.0\bin\MSBuild.exe` failed with exit code: 1

    at ModuleRebuilder.rebuildNodeGypModule (C:\Users\rannug\Documents\work\local-configurator\node_modules\electron-rebuild\lib\src\module-rebuilder.js:193:19)
    at processTicksAndRejections (internal/process/task_queues.js:95:5)
    at async Rebuilder.rebuildModuleAt (C:\Users\rannug\Documents\work\local-configurator\node_modules\electron-rebuild\lib\src\rebuild.js:190:9)
    at async Rebuilder.rebuild (C:\Users\rannug\Documents\work\local-configurator\node_modules\electron-rebuild\lib\src\rebuild.js:152:17)
    at async C:\Users\rannug\Documents\work\local-configurator\node_modules\electron-rebuild\lib\src\cli.js:146:9
PS C:\Users\rannug\Documents\work\local-configurator>
```
Any help or suggestions of where to look would be much appreciated

gunnarSensus, 2021-07-20:
im wondering if because of this log:
```
⠏ Building module: bindings, Completed: 0gyp info find VS using VS2015 (14.0) found at:
```
that somehow it is using the 2015 version of VS build tools when i have it explicitly set in my .npmrc to use 2019

GazHank, 2021-07-20:
Hi @gunnarSensus

are you still facing issues with https://github.com/serialport/node-serialport/issues/2299 or am I seeing your comments out of sequence?

If updating to VS2017 / 2019 worked per your comments on https://github.com/serialport/node-serialport/issues/2299 then it looks like this could be the fix needed (in combination with updating to python  3)

ps - at the moment the node-serialport docs have links to sources such as node-gyp and electron-rebuild in part so that if they change their dependencies it doesnt render the doc out of date.  However given a few people are hitting this sort of issue, I wonder if it would be better for everyone if we just included the info rather than a link.  @reconbot do you have a preference for how you would like to handle this?  Could the new discussions functionality in github help this sort of troubleshooting?

gunnarSensus, 2021-07-20:
i uninstalled my 2015 ms build tools install and am now getting the following error:
```
PS C:\Users\rannug\Documents\work\local-configurator> .\node_modules\.bin\electron-rebuild
⠼ Building module: bindings, Completed: 0gyp info find Python using Python version 3.9.6 found at "C:\Users\rannug\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\python.exe"
⠋ Building module: bindings, Completed: 0gyp ERR! find VS
gyp ERR! find VS msvs_version was set from command line or npm config
gyp ERR! find VS - looking for Visual Studio version 2015
gyp ERR! find VS VCINSTALLDIR not set, not running in VS Command Prompt
gyp ERR! find VS checking VS2019 (16.10.31424.327) found at:
gyp ERR! find VS "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community"
gyp ERR! find VS - found "Visual Studio C++ core features"
gyp ERR! find VS - found VC++ toolset: v142
gyp ERR! find VS - found Windows SDK: 10.0.19041.0
gyp ERR! find VS - msvs_version does not match this version
gyp ERR! find VS checking VS2017 (15.9.28307.905) found at:
gyp ERR! find VS "C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools"
gyp ERR! find VS - found "Visual Studio C++ core features"
gyp ERR! find VS - found VC++ toolset: v141
gyp ERR! find VS - found Windows SDK: 10.0.17763.0
gyp ERR! find VS - msvs_version does not match this version
gyp ERR! find VS could not find a version of Visual Studio 2017 or newer to use
gyp ERR! find VS looking for Visual Studio 2015
gyp ERR! find VS - not found
gyp ERR! find VS not looking for VS2013 as it is only supported up to Node.js 8
gyp ERR! find VS
gyp ERR! find VS valid versions for msvs_version:
gyp ERR! find VS - "2019"
gyp ERR! find VS - "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community"
gyp ERR! find VS - "2017"
gyp ERR! find VS - "C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools"
gyp ERR! find VS
gyp ERR! find VS **************************************************************
gyp ERR! find VS You need to install the latest version of Visual Studio
gyp ERR! find VS including the "Desktop development with C++" workload.
gyp ERR! find VS For more information consult the documentation at:
gyp ERR! find VS https://github.com/nodejs/node-gyp#on-windows
gyp ERR! find VS **************************************************************
gyp ERR! find VS
✖ Rebuild Failed

An unhandled error occurred inside electron-rebuild
node-gyp failed to rebuild 'C:\Users\rannug\Documents\work\local-configurator\node_modules\@serialport\bindings'.
Error: Could not find any Visual Studio installation to use

Error: node-gyp failed to rebuild 'C:\Users\rannug\Documents\work\local-configurator\node_modules\@serialport\bindings'.
Error: Could not find any Visual Studio installation to use

    at ModuleRebuilder.rebuildNodeGypModule (C:\Users\rannug\Documents\work\local-configurator\node_modules\electron-rebuild\lib\src\module-rebuilder.js:193:19)
    at processTicksAndRejections (internal/process/task_queues.js:95:5)
    at async Rebuilder.rebuildModuleAt (C:\Users\rannug\Documents\work\local-configurator\node_modules\electron-rebuild\lib\src\rebuild.js:190:9)
    at async Rebuilder.rebuild (C:\Users\rannug\Documents\work\local-configurator\node_modules\electron-rebuild\lib\src\rebuild.js:152:17)
    at async C:\Users\rannug\Documents\work\local-configurator\node_modules\electron-rebuild\lib\src\cli.js:146:9
```
from the output it looks like electron-rebuild isnt finding my 2019 or 2017 visual studio installs somehow, even though it can install/rebuild electron-serialport just fine.....

GazHank, 2021-07-20:
I know this isn't ideal, but could you try using the ["visual studio command prompt"](https://docs.microsoft.com/en-us/visualstudio/ide/reference/command-prompt-powershell?view=vs-2019) to see if that solves this issue?

if this works then it would be because your environment variables arent setup .  I can help you set those up, but I want to make sure that is the root cause first.  _Unfortunately MS VS stopped updating the environment variables in recent version in an attempt to allow you to have multiple build versions on a single machine without conflict._

gunnarSensus, 2021-07-20:
it doesnt seem to have made a difference:
```
**********************************************************************
** Visual Studio 2019 Developer PowerShell v16.10.3
** Copyright (c) 2021 Microsoft Corporation
**********************************************************************
PS C:\Users\rannug\Documents\work\local-configurator> .\node_modules\.bin\electron-rebuild
\ Building module: bindings, Completed: 0gyp info find Python using Python version 3.9.6 found at "C:\Users\rannug\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\python.exe"
\ Building module: bindings, Completed: 0gyp ERR! find VS
gyp ERR! find VS msvs_version was set from command line or npm config
gyp ERR! find VS - looking for Visual Studio version 2015
gyp ERR! find VS running in VS Command Prompt, installation path is:
gyp ERR! find VS "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community"
gyp ERR! find VS - will only use this version
gyp ERR! find VS checking VS2019 (16.10.31424.327) found at:
gyp ERR! find VS "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community"
gyp ERR! find VS - found "Visual Studio C++ core features"
gyp ERR! find VS - found VC++ toolset: v142
gyp ERR! find VS - found Windows SDK: 10.0.19041.0
gyp ERR! find VS - msvs_version does not match this version
gyp ERR! find VS checking VS2017 (15.9.28307.905) found at:
gyp ERR! find VS "C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools"
gyp ERR! find VS - found "Visual Studio C++ core features"
gyp ERR! find VS - found VC++ toolset: v141
gyp ERR! find VS - found Windows SDK: 10.0.17763.0
gyp ERR! find VS - msvs_version does not match this version
gyp ERR! find VS could not find a version of Visual Studio 2017 or newer to use
gyp ERR! find VS looking for Visual Studio 2015
gyp ERR! find VS - not found
gyp ERR! find VS not looking for VS2013 as it is only supported up to Node.js 8
gyp ERR! find VS msvs_version does not match this VS Command Prompt or the
gyp ERR! find VS installation cannot be used.
gyp ERR! find VS
gyp ERR! find VS **************************************************************
gyp ERR! find VS You need to install the latest version of Visual Studio
gyp ERR! find VS including the "Desktop development with C++" workload.
gyp ERR! find VS For more information consult the documentation at:
gyp ERR! find VS https://github.com/nodejs/node-gyp#on-windows
gyp ERR! find VS **************************************************************
gyp ERR! find VS
× Rebuild Failed

An unhandled error occurred inside electron-rebuild
node-gyp failed to rebuild 'C:\Users\rannug\Documents\work\local-configurator\node_modules\@serialport\bindings'.
Error: Could not find any Visual Studio installation to use

Error: node-gyp failed to rebuild 'C:\Users\rannug\Documents\work\local-configurator\node_modules\@serialport\bindings'.
Error: Could not find any Visual Studio installation to use

    at ModuleRebuilder.rebuildNodeGypModule (C:\Users\rannug\Documents\work\local-configurator\node_modules\electron-rebuild\lib\src\module-rebuilder.js:193:19)
    at processTicksAndRejections (internal/process/task_queues.js:95:5)
    at async Rebuilder.rebuildModuleAt (C:\Users\rannug\Documents\work\local-configurator\node_modules\electron-rebuild\lib\src\rebuild.js:190:9)
    at async Rebuilder.rebuild (C:\Users\rannug\Documents\work\local-configurator\node_modules\electron-rebuild\lib\src\rebuild.js:152:17)
    at async C:\Users\rannug\Documents\work\local-configurator\node_modules\electron-rebuild\lib\src\cli.js:146:9
PS C:\Users\rannug\Documents\work\local-configurator>
```
here is my .npmrc right now (excluding some certs):
```
msvs_version=2019
python=C:\Users\rannug\AppData\Local\Microsoft\WindowsApps\python3.exe
# msbuild_path=C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Current\Bin\MSBuild.ex
```
the msbuild_path was something i was trying from this thread: https://github.com/nodejs/node-gyp/issues/1753
again, i really appreciate your help so much

GazHank, 2021-07-20:
how about if you clear the MSbuild and msvs version info from the .npmrc and try the VS command prompt?

gunnarSensus, 2021-07-20:
exact same result. it still thinks i am setting msvs_version somehow
```
**********************************************************************
** Visual Studio 2019 Developer PowerShell v16.10.3
** Copyright (c) 2021 Microsoft Corporation
**********************************************************************
PS C:\Users\rannug\Documents\work\local-configurator> .\node_modules\.bin\electron-rebuild
/ Building module: bindings, Completed: 0gyp info find Python using Python version 3.9.6 found at "C:\Users\rannug\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\python.exe"
/ Building module: bindings, Completed: 0gyp ERR! find VS
gyp ERR! find VS msvs_version was set from command line or npm config
```

gunnarSensus, 2021-07-20:
im also working on getting a clean windows vm install so i can try from a slightly less janky windows install

GazHank, 2021-07-20:
that sounds like a sound plan :-) 

in the meantime you could try `npm config ls -l `to see what config it is using?

gunnarSensus, 2021-07-20:
here is full config (minus private npm info which i redacted):
```
PS C:\Users\rannug\Documents\work\local-configurator> npm config ls -l
; cli configs
long = true
metrics-registry = "https://10.20.159.179/artifactory/api/npm/npm/"
scope = ""
user-agent = "npm/6.14.13 node/v14.17.3 win32 x64"

; userconfig C:\Users\rannug\.npmrc
python = "C:\\Users\\rannug\\AppData\\Local\\Microsoft\\WindowsApps\\python3.exe"

; default values
access = null
allow-same-version = false
also = null
always-auth = false
audit = true
audit-level = "low"
auth-type = "legacy"
before = null
bin-links = true
browser = null
; ca = null (overridden)
cache = "C:\\Users\\rannug\\AppData\\Roaming\\npm-cache"
cache-lock-retries = 10
cache-lock-stale = 60000
cache-lock-wait = 10000
cache-max = null
cache-min = 10
cafile = undefined
cert = null
cidr = null
color = true
commit-hooks = true
depth = null
description = true
dev = false
dry-run = false
editor = "notepad.exe"
engine-strict = false
fetch-retries = 2
fetch-retry-factor = 10
fetch-retry-maxtimeout = 60000
fetch-retry-mintimeout = 10000
force = false
format-package-lock = true
fund = true
git = "git"
git-tag-version = true
global = false
global-style = false
globalconfig = "C:\\Program Files\\nodejs\\etc\\npmrc"
globalignorefile = "C:\\Program Files\\nodejs\\etc\\npmignore"
group = 0
ham-it-up = false
heading = "npm"
https-proxy = null
if-present = false
ignore-prepublish = false
ignore-scripts = false
init-author-email = ""
init-author-name = ""
init-author-url = ""
init-license = "ISC"
init-module = "C:\\Users\\rannug\\.npm-init.js"
init-version = "1.0.0"
json = false
key = null
legacy-bundling = false
link = false
local-address = undefined
loglevel = "notice"
logs-max = 10
; long = false (overridden)
maxsockets = 50
message = "%s"
; metrics-registry = null (overridden)
node-options = null
node-version = "14.17.3"
noproxy = null
offline = false
onload-script = null
only = null
optional = true
otp = null
package-lock = true
package-lock-only = false
parseable = false
prefer-offline = false
prefer-online = false
prefix = "C:\\Program Files\\nodejs"
preid = ""
production = false
progress = true
proxy = null
read-only = false
rebuild-bundle = true
; registry = "https://registry.npmjs.org/" (overridden)
rollback = true
save = true
save-bundle = false
save-dev = false
save-exact = false
save-optional = false
save-prefix = "^"
save-prod = false
scope = ""
script-shell = null
scripts-prepend-node-path = "warn-only"
searchexclude = null
searchlimit = 20
searchopts = ""
searchstaleness = 900
send-metrics = false
shell = "C:\\WINDOWS\\system32\\cmd.exe"
shrinkwrap = true
sign-git-commit = false
sign-git-tag = false
sso-poll-frequency = 500
sso-type = "oauth"
strict-ssl = true
tag = "latest"
tag-version-prefix = "v"
timing = false
tmp = "C:\\Users\\rannug\\AppData\\Local\\Temp"
umask = 0
unicode = false
unsafe-perm = true
update-notifier = true
usage = false
user = 0
; user-agent = "npm/{npm-version} node/{node-version} {platform} {arch} {ci}" (overridden)
userconfig = "C:\\Users\\rannug\\.npmrc"
version = false
versions = false
viewer = "browser"
```
i dont see anything referencing msvs_version

GazHank, 2021-07-20:
hmm... so:

- no command line settings
- nothing in the npmrc settings
- nothing in the global npmrc settings

can you check the environment variables?  

it should be accessible through a quick search from the start menu:
[image]

gunnarSensus, 2021-07-20:
culprit?
[image]

GazHank, 2021-07-20:
🤞🏼

gunnarSensus, 2021-07-20:
Success! it was able to re-build serialport, but then died while re-building the native xml library i use lol

GazHank, 2021-07-20:
that's good news (I think?)

@orange-beans and @hustzju please let us know if updating to python 3 and VS build tools 2017+ help resolve this for you too?

@gunnarSensus I'm not sure I can help, but feel free to @ me in your issue/ investigation into the native xml library issue, I'd be happy to try to help if I can

closed 2022-04-28 by stale[bot]