serialport/node-serialport #2195: Add version to PortInfo

phhoef, 2021-01-01:
### Summary of Problem
I am wondering, if it's possible to get the version of the device with serial port.
When a device is connected to the OS, a version number is shown in the system information.
On macOS it looks like this:
[image]

Is it possible to retrieve this version?
I was hoping, that the `PortInfo` class is holding the information after calling `serialport.list()`, but it seems, that it isn't.

reconbot, 2021-01-01:
Do you know if it provides a serial device in /dev ? It might be usb only.

phhoef, 2021-01-01:
yes. In the shown example the AVR programmer is registering two devices in /dev.
```
~ ls /dev |grep usb
cu.usbmodem001319021
cu.usbmodem001319023
tty.usbmodem001319021
tty.usbmodem001319023
```
Strictly speaking, even 4 devices: 2 tty and 2 cu

Maybe it's a bad example. I am trying to get the version number of an Arduino Due.
Unfortunately, I do not have one here right now, that's why I used the AVR programmer as example - sorry for confusion.
That said, the Arduino shows also the version in system information and registers under /dev, too.

phhoef, 2021-01-01:
Probably you're right. I am no expert, but as far as I understand is it the `bcdDevice` version.
It seems, that it's only available for USB devices. Therefore it is not available through serial port, right?

reconbot, 2021-01-01:
On osx we use [`findModems`](https://developer.apple.com/documentation/iokit/communicating_with_a_modem_on_a_serial_port) to find available ports. 

https://github.com/serialport/node-serialport/blob/master/packages/bindings/src/darwin_list.cpp#L121-L138

So we don't have a lot of control here.

phhoef, 2021-01-01:
Thanks for the insights. I looked through the IOKit documentation and I think I found the key for the version number. 
from my point of view it is `kUSBDeviceReleaseNumber`. 
https://developer.apple.com/documentation/iokit/kusbdevicereleasenumber

is it possible to read this value and attach is to the `PortInfo`?
Not sure how it would work on windows/linux.