serialport/node-serialport #2199: Mock binding only working in sync enviroment

WarNote, 2021-01-08:
### Summary of Problem

- What are you trying to do?
I'm trying to make a loop of sending data, from an instantiated MockBinding class, with emitData method.

- What happens?
Only works in sync, when the process begin looping with async, it stops working. I've tried process.nextTick, setInterval, async iterations, async event loops, intermediate event to handle sync and async separately but none worked.

What i've tried and worked is to fire the event inside the port event "data" listener, but that would break the project and i would probably drop the mock binding.

- What should have happened?
It should've logged to console from the events that i was emitting.

### Code to Reproduce the Issue

```js
const SerialPort = require('serialport');
const { EventEmitter } = require('events');
const MockBinding = require('@serialport/binding-mock');

const binding = new MockBinding({});
SerialPort.Binding = MockBinding;

MockBinding.createPort('/dev/ROBOT', { vendorId: 'vendorId', productId: 'productId' });

const event = new EventEmitter();

const emitData = () => binding.emitData('event testing data');

event.addListener('test', emitData);

const tickLoop = (callback) => {
  callback();
  process.nextTick(tickLoop, callback);
}

binding.open('/dev/ROBOT', { baudRate: 15000 }).then(() => {
  setInterval(emitData, 200);
  // setInterval(() => event.emit('test'), 200);
  // tickLoop(emitData);
  // tickLoop(() => binding.emitData('testing data'));
});

const port = new SerialPort('/dev/ROBOT', { baudRate: 15000 });
port.addListener('data', (data) => {
  console.log(data.toString());
});
```
*Remove the comments one by one to try the alternatives that i tought it would suffice.

### Versions, Operating System and Hardware

- SerialPort@9.0.3
- @serialport/binding-mock@9.0.2
- Node.js v12.18.1 / 14.15.1
- Windows
- Amd ryzen 5 1600

davecranwell, 2021-04-23:
Did you get anywhere with this? I'm stuck here too. Feel like it's something obvious I'm not seeing, but no lightbulb moment yet.

AyAyEm, 2021-04-23:
> Did you get anywhere with this? I'm stuck here too. Feel like it's something obvious I'm not seeing, but no lightbulb moment yet.

Nope i used this:

> What i've tried and worked is to fire the event inside the port event "data" listener, but that would break the project and i would probably drop the mock binding.

But it just worked because i was using OO with event driven paradigm, i don't believe i would be able to do it using imperative, procedural or functional.

Here the project that i'm working with and the solution i used to the problem:
https://github.com/AyAyEm/node-balances/blob/main/src/structures/balanceEmulator.ts

You can find how i dit it in the connect method.

_Disclaimer: It was a pain in the ass to work with mock binding because there's no typing for it, and it lacks a lot in documentation. Then the only solution was to read code which was a pain too, i never get used to read OO._

reconbot, 2022-04-16:
For the record this improved a bit with v10, types for everything and it's a bit easier to use and inject.

closed 2022-04-16 by reconbot