serialport/node-serialport #2200: Cannot run after buiding in electron

ArcherGu, 2021-01-10:
### Summary of Problem

- What are you trying to do?
I'm using serialport in electron, which works well in developer mode (`npm run electron:serve`) after electron-rebuild.
 
- What happens?
When I run `npm run electron:build`, it builds well，but it cant run(Without any errors,  just can't open .exe).

- What should have happened?
It should run well just like in dev mode.

I've pinpointed that this problem is caused by serialport, and everything was fine when I stopped using it in my project.

In fact, I built my project in the same way about a month ago, and it worked well, but recently I added something new to my project and tried to recompile it, and then found out about the problem.

### Code to Reproduce the Issue

```js
// serialport.js
const SerialPort = require('serialport');
const port = new SerialPort('COM3', {
    baudRate: 57600
});

//vue.config.js
module.exports = {
    pluginOptions: {
        electronBuilder: {
            builderOptions: {
                nsis: {
                    oneClick: false,
                    allowElevation: true,
                    allowToChangeInstallationDirectory: true,
                    createDesktopShortcut: true,
                    createStartMenuShortcut: true,
                }
            },
            externals: ['serialport']
        }
    }
};
```

### Versions, Operating System and Hardware

- SerialPort@9.0.2
- Node.js v14.15.1
- Electron v9.4.0
- vue-cli-plugin-electron-builder ~2.0.0-rc.5
- Windows10
- I7-9700F

ArcherGu, 2021-01-11:
Hi, it looks like I've found the problem, serialport doesn't have any problems, it's on [electron-builder](https://github.com/electron-userland/electron-builder).

When I finish electron-rebuild, electron-builder rebuilds the serialport again(`npm rebuild`) when I execute `npm run electron:build`.

so I added:
```js
npmRebuild: false
```
Everything's fine.

Next I should go to [electron-builder](https://github.com/electron-userland/electron-builder) to see some information.

yetianyue, 2021-01-21:
@ArcherGu  thinks

closed 2021-01-21 by ArcherGu