serialport/node-serialport #2201: Exec format error on running Node.js app using Dockerfile + Raspberry Pi 3 + balenaCloud

dakshshah96, 2021-01-11:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?

Use `node-serialport` as a dependency in my Node.js application for a Raspberry Pi 3.

My Dockerfile looks like this:

```
# Base image for node 14.15.1 on raspberry pi 3
FROM balenalib/raspberrypi3-alpine-node:14.15.1-run

# Install packages
RUN install_packages make gcc g++ python2 linux-headers udev

# Defines our working directory in container
WORKDIR /usr/src/app

# Copies the package.json first for better cache on later pushes
COPY package.json ./package.json

# This install npm dependencies on the balena build server,
# making sure to clean up the artifacts it creates in order to reduce the image size
RUN JOBS=MAX npm install --production --unsafe-perm --build-from-source \
  && npm prune --production \
  && npm cache clean --force \
  && rm -rf /tmp/*

# This will copy all files in our root to the working  directory in the container
COPY . ./

# Enable udevd so that plugged dynamic hardware devices show up in our container
ENV UDEV=1

# Run this when container starts up on the device
CMD ["npm", "start"]
```

- What happens? 

The Node.js app starts crashing as soon as I run `npm start` in the Dockerfile

```
uncaughtException: Error loading shared library /usr/src/app/node_modules/@serialport/bindings/build/Release/bindings.node: Exec format error
    Error: Error loading shared library /usr/src/app/node_modules/@serialport/bindings/build/Release/bindings.node: Exec format error
        at Object.Module._extensions..node (internal/modules/cjs/loader.js:1122:18)
        at Module.load (internal/modules/cjs/loader.js:928:32)
        at Function.Module._load (internal/modules/cjs/loader.js:769:14)
        at Module.require (internal/modules/cjs/loader.js:952:19)
        at require (internal/modules/cjs/helpers.js:88:18)
        at bindings (/usr/src/app/node_modules/bindings/bindings.js:112:48)
        at Object.<anonymous> (/usr/src/app/node_modules/@serialport/bindings/lib/linux.js:2:36)
        at Module._compile (internal/modules/cjs/loader.js:1063:30)
        at Object.Module._extensions..js (internal/modules/cjs/loader.js:1092:10)
        at Module.load (internal/modules/cjs/loader.js:928:32)
```

### Versions, Operating System and Hardware

- SerialPort@? 9.0.4
- Node.js v? 14.15.1
- Windows? Linux? Mac? Dockerfile using Alpine Node 14.15.1
- Hardware and chipset? (Prolific/FTDI/Other) Raspberry Pi 3

gigoh, 2021-01-29:
It seems to be incompatible binary. I guess that the image was built on the desktop but you are trying to use it on the pi.

The images was built on the pi, however, I suspect some library (dependency) could be built on the different kernel. AFAIK, Alpine provides `aarch64`, `armhf`, `armv7` dist for the pi.

How about try to build `@serialport/bindings` inside the docker environment in manually?

GazHank, 2021-07-05:
Hi @dakshshah96 did rebuilding from source solve your issue?

closed 2022-04-28 by stale[bot]