serialport/node-serialport #2207: apps crash on launch using Node JS 15 with serialport

sceptre357, 2021-01-24:
I've deleted my node_modules directory after upgrading to NodeJS 15.6 (from Node 14). i then installed serialport into my app directory using "npm i serialport" , a new node_modules folder was created but now i can't load my apps. 

says:
/apps/appname/node_modules/bindings/bindings.js:135
  throw err;
  ^
Error: Could not locate the bindings file. Tried:

is there something missing from NPM? do i need to install something else?

Environment: (multiple devices dont work)
LMDE 4
Debian 10.7
Armbian 20.11.6 Buster

reconbot, 2021-01-24:
Did you skip the install scripts?

ngvcanh, 2021-02-05:
Let's try rebuild packages:

```
npm i electron-rebuild --save-dev
./node_modules/.bin/electron-rebuild
```

closed 2022-04-28 by stale[bot]