serialport/node-serialport #2208: SerialPort.write is not working !

VNedu, 2021-01-24:
### Summary of Problem
hey! I'm trying to send a string data to Arduino UNO, but Serialport.write is not working  :( ?
### Code to Reproduce the Issue

```js
const SerialPort = require('serialport');
const Readline = require('@serialport/parser-readline');
const port = new SerialPort('COM3',{ boundrate : 9600 }  );

const parser = new Readline();
port.pipe(parser);

// read data  
parser.on('data', console.log);

 // write data 
 setTimeout(function() {
    port.write("Serialport-data\n")
  },2000);
```
### Arduino code 
```c++
String incomingString  ;
void setup() {
     Serial.begin(9600);
       }

void loop() { 
     if (Serial.available() >= 0 ){
        Serial.println("received"+ incomingString);
        incomingString = Serial.readString();
      delay(2000);
  }
}

```

### Versions, Operating System and Hardware

- SerialPort@ 9.x.x
- Node.js v14.15.4
- Windows 10

reconbot, 2021-01-24:
You misspelled baudRate, what are you seeing? Any errors?
-- 
---
Francis Gulotta
wizard@roborooter.com

VNedu, 2021-01-24:
No, there are no errors, it is reading data from SerialPort but not writing any data on SerialPort.

reconbot, 2021-01-25:
Change your set timeout to a setInterval? I often see arduinos take a few moments to boot up and loose serial data. It's usually less than 2 seconds however.

There's a "ready parser" that waits for the arduino to say it's ready and emits an event for it. An approach like this if often used if there's no dtr lines (there isn't for arduino).

You can also try debug mode to see if that tells you anything. See serialport.io for instructions

apiel, 2021-02-15:
I am having the same problem. When I use the serial monitor from arduino, writing work well. But if I use SerialPort.write it doesn't work.
But if I print something in the arduino code, then writing from SerialPort is working -.- Kind of weird.
```C
    if (Serial.available()) {
        int data = Serial.read();
        Serial.print("I received: ");
        Serial.println(data);
    } else {
        Serial.print(".");
    }
```

apiel, 2021-02-16:
There might be some binding issue.The following code is not working properly:
```js
const easymidi = require('easymidi');
const SerialPort = require('serialport');

const port = new SerialPort('/dev/ttyUSB0', {
    baudRate: 115200,
});

port.on('error', function (err) {
    console.log('Error: ', err.message);
});

port.on('data', (data) => {
    process.stdout.write(data.toString());
});

const inputs = easymidi.getInputs();
const input = new easymidi.Input(inputs[1]);
input.on('message', ({ controller, value, _type, note }) => {
     port.write("hello serial\n");
});
```
But if I had this at the end of the code:
```js
setInterval(() => port.write("\n"), 10);
```
All my message are received. So I endup to do:
```js
const easymidi = require('easymidi');
const SerialPort = require('serialport');

const port = new SerialPort('/dev/ttyUSB0', {
    baudRate: 115200,
});

port.on('error', function (err) {
    console.log('Error: ', err.message);
});

port.on('data', (data) => {
    process.stdout.write(data.toString());
});

let msg;
setInterval(() => {
  if (msg) {
    port.write(msg);
    msg = null;
  }
}, 10);

const inputs = easymidi.getInputs();
const input = new easymidi.Input(inputs[1]);
input.on('message', ({ controller, value, _type, note }) => {
     msg = "hello serial\n";
});
```
Then the message arrive to my arduino but this is super hacky!! I can't find out where is the problem. I tried to write the code in many different way without success, unless to use this ugly hack.

closed 2022-04-28 by stale[bot]