serialport/node-serialport #2209: Missing serialport dependency

Aruseli, 2021-01-26:
What am I doing wrong?
serialport@9.0.6
Node.js v12.18.2
MacOS Catalina 10.15.7

When trying to run the `node blink.js`, the `Missing serialport dependency` error is thrown
When trying to install serialport, the following errors occur:

```
@serialport/bindings@8.0.8 install /Users/ako/Desktop/arduino : esp/NodeMCU/apollo/node_modules/firmata/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

sh: prebuild-install: command not found
No receipt for 'com.apple.pkg.CLTools_Executables' found at '/'.

No receipt for 'com.apple.pkg.DeveloperToolsCLILeo' found at '/'.

No receipt for 'com.apple.pkg.DeveloperToolsCLI' found at '/'.

gyp: No Xcode or CLT version detected!
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/Users/ako/.nvm/versions/node/v12.18.2/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:351:16)
gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:275:12)
gyp ERR! System Darwin 19.6.0
gyp ERR! command "/Users/ako/.nvm/versions/node/v12.18.2/bin/node" "/Users/ako/.nvm/versions/node/v12.18.2/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/ako/Desktop/arduino : esp/NodeMCU/apollo/node_modules/firmata/node_modules/@serialport/bindings
gyp ERR! node -v v12.18.2
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok 
npm WARN apollo@1.0.0 No description
npm WARN apollo@1.0.0 No repository field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@8.0.8 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the @serialport/bindings@8.0.8 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/ako/.npm/_logs/2021-01-26T16_32_28_807Z-debug.log
```

yuzi40277738, 2021-01-28:
We are using Node.js 12.18.3, Chromium 87.0.4280.141, Electron 11.2.0, and Serialport 9.0.6 ---is OK

kakashy, 2021-04-03:
Did you find a work around to this?

Aruseli, 2021-04-12:
@kakashy yes, I reinstalled the XCode Command Line Tools

GazHank, 2021-07-10:
This looks to have been resolved, please should if any extra help or support is needed

closed 2021-07-10 by GazHank