serialport/node-serialport #2214: Windows 10 64-bit electron 9.4.2 "is not a valid win32 Application"

karlsnyder0, 2021-02-03:
I've had an issue with the Windows 10 64 bit binary for electron 9.4.2.  The error I was getting is "is not a valid win32 Application"

I've manually compiled a new binary and overrode its download using .npmrc

`@serialport/bindings_binary_host=https://myawss3bucket.s3.amazonaws.com/prebuild-install`

See attached-

[bindings-v9.0.4-electron-v80-win32-x64.tar.gz](https://github.com/serialport/node-serialport/files/5915363/bindings-v9.0.4-electron-v80-win32-x64.tar.gz)

simeon-iotsyst, 2021-02-04:
I have exactly the same issue!

GazHank, 2021-07-05:
I believe this relates to some historic issues with the prebuilds, are you facing any issues with the more recent releases?

closed 2022-04-28 by stale[bot]

Marlonzao, 2023-06-21:
> I believe this relates to some historic issues with the prebuilds, are you facing any issues with the more recent releases?

Yes, [this issue](https://github.com/serialport/node-serialport/issues/2619)