serialport/node-serialport #2215: "_LIBCPP_TRIVIAL_PAIR_COPY_CTOR" is no longer supported

amentain, 2021-02-06:
FreeBSD 12.2-RELEASE-p3 GENERIC  amd64

```bash
git clone https://github.com/serialport/node-serialport.git
cd node-serialport
npm install
```

```log
> core-js-pure@3.6.5 postinstall /usr/home/hb/node-serialport/node_modules/core-js-pure
> node -e "try{require('./postinstall')}catch(e){}"

Thank you for using core-js ( https://github.com/zloirock/core-js ) for polyfilling JavaScript standard library!

The project needs your help! Please consider supporting of core-js on Open Collective or Patreon:
> https://opencollective.com/core-js
> https://www.patreon.com/zloirock

Also, the author of core-js ( https://github.com/zloirock ) is looking for a good job -)

> serialport-monorepo@0.0.0 postinstall /usr/home/hb/node-serialport
> lerna bootstrap --no-ci

lerna notice cli v3.22.1
lerna info Bootstrapping 17 packages
lerna info Installing external dependencies
lerna ERR! npm install exited 1 in '@serialport/bindings'
lerna ERR! npm install stdout:

> @serialport/bindings@9.0.4 install /usr/home/hb/node-serialport/packages/bindings
> node-gyp rebuild

gmake: Entering directory '/usr/home/hb/node-serialport/packages/bindings/build'
  CXX(target) Release/obj.target/bindings/src/serialport.o
gmake: Leaving directory '/usr/home/hb/node-serialport/packages/bindings/build'

lerna ERR! npm install stderr:
In file included from ../src/serialport.cpp:1:
In file included from ../src/./serialport.h:3:
In file included from /usr/include/c++/v1/stdio.h:101:
/usr/include/c++/v1/__config:122:2: error: "_LIBCPP_TRIVIAL_PAIR_COPY_CTOR" is no longer supported.        use _LIBCPP_DEPRECATED_ABI_DISABLE_PAIR_TRIVIAL_COPY_CTOR instead
#error "_LIBCPP_TRIVIAL_PAIR_COPY_CTOR" is no longer supported. \
 ^
1 error generated.
gmake: *** [bindings.target.mk:105: Release/obj.target/bindings/src/serialport.o] Error 1
gyp ERR! build error
gyp ERR! stack Error: `gmake` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:198:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:12)
gyp ERR! System FreeBSD 12.2-RELEASE-p3
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /usr/home/hb/node-serialport/packages/bindings
gyp ERR! node -v v10.23.2
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@9.0.4 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the @serialport/bindings@9.0.4 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/hb/.npm/_logs/2021-02-06T12_09_07_152Z-debug.log

lerna ERR! npm install exited 1 in '@serialport/bindings'
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport-monorepo@0.0.0 postinstall: `lerna bootstrap --no-ci`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport-monorepo@0.0.0 postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
```

reconbot, 2021-02-06:
What version of node?

amentain, 2021-02-06:
v10.23.2

reconbot, 2021-02-06:
Looks like some sort of FreeBSD c++ difference. http://clang-developers.42468.n3.nabble.com/libcxx-Removing-LIBCPP-TRIVIAL-PAIR-COPY-CTOR-option-td4052064.html

I don't know how to approach solving it. We support Linux and OS X and even if you can get it to compile I don't know if all operations will work.

closed 2021-07-05 by reconbot