serialport/node-serialport #2217: Writing to UART not working on Electron APPX / UWP ( GetOverlappedResult)

mberrg, 2021-02-11:
### Summary of Problem

I'm using serialport inside an Electron application published as an Appx application (this is the same as a UWP application).

### steps
* List serial ports  - works
* Connect a serial port - works
* Write - fails
```js
Writing to COM port (GetOverlappedResult): Invalid handle

```

The exact same code works when I build run electron in dev mode or normal windows package. The error only surfaces when the electron application is packed in an appx / UWP app. Any suggestions?

####
NB: I've tried to add extra access in the app manifest for appx, but no change.
```xml
<DeviceCapability Name="serialcommunication">
      <Device Id="any">
        <Function Type="name:serialPort" />
      </Device>
</DeviceCapability>
```

GazHank, 2021-07-05:
Hi @mberrg did you manage to resolve this issue?

There are some troubleshooting steps @ https://docs.microsoft.com/en-us/uwp/api/windows.devices.serialcommunication?view=winrt-20348

I appologise if you have already reviewed those.

Out of interest, are you using [electron-builder](https://www.electron.build/) or something else?

bergmorten, 2021-07-05:
Hi, I've moved on and use web serial instead of node serial. So I do not know if this error still exists.

GazHank, 2021-07-05:
@bergmorten thanks for the speedy response.  I'll close this for now, but happy to investigate further if anyone else is facting this issue

closed 2021-07-05 by GazHank

491239, 2025-03-05:
> [@bergmorten](https://github.com/bergmorten) thanks for the speedy response. I'll close this for now, but happy to investigate further if anyone else is facting this issue

I face this  issue now, when i was developing, all things ok, but when i pack electron app to misx and upload to microsoft store, and download it, try to use, it's not work. Same as this issue: 
List port is ok, open is ok, when i write ,Serial operation error: Writing to COMport (GetOverlappedResult): Invalid handle