serialport/node-serialport #2218: What is recommended method to code tightly coupled RX/TX logic (parser should not TX, right)?

raintonr, 2021-02-11:
I see in the doco it says...

> Parsers are collections of transform streams to processes incoming data

... which makes it is pretty clear that a parser is *only* for incoming processing. So what about protocols where TX messages must be sent a specific time after one is received? Ie. an outbound message must be sent in a specific time slot which is actually defined by the protocol and as it's relative to a received message, only known to the parser.

I am trying to (in fact, have, see https://github.com/raintonr/hormann-hcp for currently experimental - and thus rather ugly - code) create an interface for Hormann HCP protocol which is one such example of this kind of protocol.

The protocol specifies exact timing for transmitting responses which must be sent 3-10ms (I think!) after the first byte of an incoming message was received (RX bytes).

Currently I have implemented in the parser, a timestamp variable that can be found with a 'get' method which defines the time of an incoming message. The caller reads this and times a response (TX bytes) appropriately.

To me this doesn't seem appropriate, as it splits the logic of the protocol across two code modules (the parser and the caller). Let's not even mention they both need to share a similar CRC computation method and cyclic message 'counter' which are also part of the protocol.

Is there a recommended way of coding tightly coupled TX/RX logic like this? According to the doco, a parser is not it, so is there another way?

It's not that what I have done doesn't work, but it seems rather inelegant is all.

reconbot, 2021-02-23:
Well, you can use a [`duplex stream`](https://nodejs.org/api/stream.html#stream_class_stream_duplex) for it. But there isn't a wrong way to do it if it works.

raintonr, 2021-02-27:
> Well, you can use a [`duplex stream`](https://nodejs.org/api/stream.html#stream_class_stream_duplex) for it. But there isn't a wrong way to do it if it works.

Thanks for replying. So no problem then, think putting everything in the same file and doing that is the way to go in this case.

closed 2021-02-27 by raintonr