serialport/node-serialport #2219: Getting Errors after installing serial port in Angular Project

NayanMiyatra, 2021-02-12:
I'm trying to implement data reading through a serial port in my Angular project (using electron), but after installing the serial-port package I'm getting errors of 'Module not found'.

The followings are my reference for the project

- https://stackoverflow.com/questions/48252859/angular5-electron-serialport-support
- https://www.npmjs.com/package/serialport

electron service code

```
import { Injectable } from '@angular/core';
import * as SerialPort from 'serialport';

@Injectable({
  providedIn: 'root'
})

export class ElectronService {
  serialPort: typeof SerialPort;
  constructor() {
    this.serialPort = require('serialport');
  }
}
```
My type-script file

```
import { Component } from '@angular/core';
import * as SerialPort from 'serialport';
import { ElectronService } from './electron.service';

@Component({
  selector: 'app-root',
  templateUrl: './app.component.html',
  styleUrls: ['./app.component.scss']
})

export class AppComponent {
  title = 'serial-port';
  serialPort: typeof SerialPort;

  constructor(private electron: ElectronService) {

  }
  ngOnInit(): void {
    this.electron.serialPort.list().then((ports: any) => {
      {
        console.log("true")
      }
    }).catch((err: any) => {
      {
        console.log("Error" + err)
      }
    });
    console.log(this.serialPort);
  }

}
```
The error I'm getting
[image]

SumitRajpal, 2021-05-02:
@NayanMiyatra  have you solve this error? particularly I am working on the web project.
https://stackoverflow.com/questions/48252859/angular5-electron-serialport-support I got only this ref link.

GazHank, 2021-07-05:
Hi, have you had any luck solving these issues?

I'm not familiar with Angular myself, so wondered if you had any joy reviewing other projects that use the same stack, either to check if they work in your build environment or to review their code for pointers.  For example, https://github.com/yu4321/Angular-Electron-SerialPort_ThermalPrinterExample looks to have been relateively recently maintained so might be a good starting point?

closed 2022-04-28 by stale[bot]