serialport/node-serialport #2220: Opening COM3: Access denied

Kinqdos, 2021-02-16:
### Summary of Problem

(Please answer all 3)

- What are you trying to do? => Display data send from Arduino via serial connection on Electron programm
- What happens? Getting error (node:724) UnhandledPromiseRejectionWarning: Error: Opening COM3: Access denied
- What should have happened? No error

### Code to Reproduce the Issue

```js
const serialport = new SerialPort("COM3", {baudRate: 9600))
serialport.on("open", () => {
    console.log("open")
})
serialport.on("readable", () => {
            console.log(serialport.read())
})
```

### Versions, Operating System and Hardware

- SerialPort@newest
- Node.js v14.15.5
- Windows? Linux? Mac? Testing on Windows 10 => Purpose RasperryPi
- Hardware and chipset? (Prolific/FTDI/Other)

Im trying to show data in an electron application, send from arduino via serial connection.
First issue was the node module version, but fixed it with electron-rebuild.
After the rebuild electron is starting but im getting the folling error:
```
(node:3528) UnhandledPromiseRejectionWarning: Error: Opening COM3: Access denied
(node:3528) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch()
. To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:3528) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

After the error it still logs "open" and every second "null" (serialport.read()).
What can I do to fix the problem ?

YounesTea, 2021-02-19:
The error says you're not handling some Promise in your code; is your code open source?

Kinqdos, 2021-02-19:
Sure, the electron entry is in app/app.ts and the serialport stuff is at app/ControllerConenction.ts
https://github.com/Kinqdos/DickesWomo

YounesTea, 2021-02-19:
I can't run your code via `npm run start`. Can you provide a working Docker image?

Kinqdos, 2021-02-19:
Sry im not really familiar with docker but i pushed a new version.
You just have to run ```npm install``` compile all Typescript files and run ```npm run start```.

Kinqdos, 2021-02-20:
Hey found the error by myself. Me dumpass created a new serialport on the same port twice.
Thank you for your effort !

closed 2021-02-20 by Kinqdos

reconbot, 2021-02-20:
Happens to everyone no worries

YounesTea, 2021-02-20:
> Hey found the error by myself. Me dumpass created a new serialport on the same port twice.
> Thank you for your effort !

Just saw your reply. I'm glad you found the root cause!