serialport/node-serialport #2224: Can't read back FTDI USB serial number in docker container

igd-tom, 2021-02-24:
### Summary of Problem

**What are you trying to do?** 
Use node-serial inside node:14-apline docker container to read the serial number of an FTDI RS232 serial device.

**What happens?**
I mount the dev folder and run container in privileged mode and can read and write to the FTDI serial device but when I use  'SerialPort.list()' to read the USB details, the Vendor ID, Product ID, Serial Number etc.  are undefined. The only defined field is the path which correctly displays 'ttyUSB0'.

**What should have happened?**
Running this inside my host machine (Ubuntu 18.04.5 LTS) I can read back all the serial device details including Vendor ID, Product ID and Serial Number. 

### Code to Reproduce the Issue

See my repo: [](https://github.com/igd-tom/simple-serial-node-docker)

After cloning the repo, the following commands are performed from the root directory:

```
docker build -t simple_serial_node .
docker run -v /dev:/dev simple_serial_node

```

### Versions, Operating System and Hardware

- SerialPort@: 9.0.7
- Node.js v14
- Ubuntu 18.04.5 LTS
- Hardware and chipset? FTDI

GazHank, 2021-07-05:
Hi @igd-tom

are you still facing issues?

it looks like ther are a few ways to setup the access to the device, and some of these should avoid the use of --privileged ; however I've not been able to check if they change the data available as I dont have access to my Linux machine at the moment, and windows just behaves differently with docker.

some potentially useful resources:

https://stackoverflow.com/questions/24225647/docker-a-way-to-give-access-to-a-host-usb-or-serial-device

http://marc.merlins.org/perso/linux/post_2018-12-20_Accessing-USB-Devices-In-Docker-_ttyUSB0_-dev-bus-usb-_-for-fastboot_-adb_-without-using-privileged.html

igd-tom, 2021-07-14:
Thanks for your response @GazHank. I managed to resolve it by tackling the problem in a different way. I used UDEV rules on the host OS to read back the FTDI chipset details and map to a specific symlink name which could then be accessed from the docker container.

GazHank, 2021-07-17:
Thanks @igd-tom since you have solved this, I'll close the issue.  Do let me know if you think any of the steps you took to solve this could be helpful for inclusion in the docs

closed 2021-07-17 by GazHank