serialport/node-serialport #2227: serialport - Electron Binding Issue

Sourabh-Zutti, 2021-03-06:
I am not able to find proper bindings for serialport with electron version > 4.

Error Message :

..\src\serialport.cpp(329): error C2661: 'v8::Object::Set': no overloaded function takes 2 arguments [E:\LastestUpdated
Workspace\workspace\profitguru\profitGuruDAPP\node_modules\@serialport\bindings\build\bindings.vcxproj]
..\src\serialport.cpp(330): error C2661: 'v8::Object::Set': no overloaded function takes 2 arguments [E:\LastestUpdated
Workspace\workspace\profitguru\profitGuruDAPP\node_modules\@serialport\bindings\build\bindings.vcxproj]
..\src\serialport.cpp(331): error C2661: 'v8::Object::Set': no overloaded function takes 2 arguments [E:\LastestUpdated
Workspace\workspace\profitguru\profitGuruDAPP\node_modules\@serialport\bindings\build\bindings.vcxproj]
..\src\serialport.cpp(378): error C2661: 'v8::Object::Set': no overloaded function takes 2 arguments [E:\LastestUpdated
Workspace\workspace\profitguru\profitGuruDAPP\node_modules\@serialport\bindings\build\bindings.vcxproj]
× Rebuild Failed

- What are you trying to do? - I am trying to find the Bindings for serialport.
- What happens? - I am getting errors, not able to get Bindings for serialport
- What should have happened? - Should get Proper Bindings 

### Versions, Operating System and Hardware

- SerialPort@7.1.5
- Node.js v10.15.3
- Windows? Linux? Mac? Windows 10
- electron@9.2.0

partap, 2021-06-30:
Me too. :(
No prebuilt bindings ever seem to be available anymore...but it *did* seem to work with electron v4

Only solution I know of is to force electron-rebuild to rebuild with the proper electron version specified on the cmdline, e.g.:
`electron-rebuild -f -v 10.15.3 -w serialport`

Sourabh-Zutti, 2021-06-30:
Hello @partap, Thanks for your sharing of Knowledge. But would like to share with you I found the bindings for the serial port for electron version 9.0.4. It seems they have recently provided the serial port bindings for the electron module. So updated electron version to 9.0.4. Now All working well!

closed 2021-06-30 by Sourabh-Zutti

partap, 2021-06-30:
I'm using serialport 9.2.0 with electron 12 (abi: 87) and the bindings _are_ supposedly there:
https://github.com/serialport/node-serialport/releases/tag/%40serialport%2Fbindings%409.2.0

But they never get downloaded by electron-builder, and electron-rebuild just builds from source

Sourabh-Zutti, 2021-06-30:
@partap You can try using the serial port version 7.1.5 and electron 9.0.4. It should work!

partap, 2021-06-30:
Are you building serialport from source or is it actually downloading the prebuilts?

I need to use electron 12 because of fixes to the video player in chrome 89...
Before that I was using electron 4...I was able to use the prebuilts to build windows and linux releases on my mac, but now I need to spin up VMs because electron-builder doesn't seem to be able to find the prebuilts and it won't cross-compile...

Maybe it's just an issue with electron-builder though.