serialport/node-serialport #2228: SerialPort.list() doesn't work anymore?

bluephoton, 2021-03-10:
### Summary of Problem
- What are you trying to do? **Listing serial ports**
- What happens?  **An error (random number each time I run the code)**
- What should have happened? **Should get the list**

### Code to Reproduce the Issue

```js
   const SerialPort = require("serialport");
   SerialPort.list().then(ports => {
            console.log("OK");
   },
   err => {
      console.log(err);
   });
```

### Versions, Operating System and Hardware

- SerialPort@?  **9.0.7**
- Node.js v?  **v14.16.0**
- Windows? Linux? Mac? **Windows 10**
- Hardware and chipset? (Prolific/FTDI/Other)

hdas2012, 2021-04-01:
I am facing same issue.

GazHank, 2021-07-05:
Hi, are you still facing issues with the latest versions?

vshtykhnov, 2021-07-07:
> Hi, are you still facing issues with the latest versions?

yes

GazHank, 2021-07-07:
Would you be able to share a sample of the error(s) you are receiving?

Could you also share some info on the devices you have connected?  (A screen print of the info from device manager or similar?)

The reason for the devices question is that there are a few historic errors on windows which seem to be related to specific devices where either windows or the device itself doesn't behave as expected

Thanks

Gaz

vshtykhnov, 2021-07-07:
I don't have this problem on version 8.0.8
Regardless of the devices that are connected, even if they do not exist at all, error when receiving ports (random number each time I run the code)

vshtykhnov, 2021-07-07:
> Would you be able to share a sample of the error(s) you are receiving?
>
> Could you also share some info on the devices you have connected? (A screen print of the info from device manager or [...]

I don't have this problem on version 8.0.8
Regardless of the devices that are connected, even if they do not exist at all, error when receiving ports (random number each time I run the code)

GazHank, 2021-07-10:
Hi @vshtykhnov 

is there any more info you can share, as I've tried downgrading to your version of node, and running exactly the same code as your example and I can't seem to create any errors:

```
PS C:\Git\example-node-serialport> node main.js
OK
PS C:\Git\example-node-serialport> node -v
v14.16.0
```
This is based on Win10 (19043.1083)

jcasp-dev-rohan, 2021-08-12:
Hi,

Does anyone have any idea why this (below) code return an empty array on a windows 10 laptop ?

```
const SerialPort = require('serialport')

SerialPort.list()
  .then((ports) => {
    console.log(ports, "s")
  })
  .catch(err => console.log(err, "e"))
```
Tech Stack:
Node: v14
Electron: v12
serialport: v9

GazHank, 2021-08-13:
@jcasp-dev-rohan since you are invoking serialport from electron have you included the [workarounds required to support electron v9+](https://serialport.io/docs/guide-installation#invoking-serialport-within-the-renderer-processes)

jcasp-dev-rohan, 2021-08-14:
@GazHank The work around is for invoking serialport from the renderer process. Since this code is invoked from the main process, it should work out of the box right ?

GazHank, 2021-08-14:
@jcasp-dev-rohan yes that's correct per the docs, but I was wondering if it was worth a try just in case... if that does resolve it then I'll update the docs to clarify

jcasp-dev-rohan, 2021-08-14:
@GazHank Okay so I made a pretty basic mistake here, I forgot to run electron-rebuild (which after running gave me error: " The syntax of the command is incorrect.").

SebCanet, 2021-09-14:
Hi,
same pb with:
- node 16.9.1
- serial 9.2.1
- windows 7 or 10, 64 bits
- electron 13.4.0
- python 3
- VS 2019
Thanks.

GazHank, 2021-09-14:
Hi @SebCanet are you using electron-rebuild or receiving any errors from the rebuild process?

SebCanet, 2021-09-14:
No error from rebuild process, here are my scripts:
"scripts": {
        "rebuild": "electron-rebuild -f -w serialport @serialport/bindings /m",
        "prepare": "electron-build-env npm rebuild",
        "start": "electron .",
        "compile": "electron-builder --win --ia32"
    },
Thanks.

GazHank, 2021-09-14:
Thanks, is there a reason for using ia32?

SebCanet, 2021-09-15:
Compatibility purpose, for education...as we have no money to change our computer, we still have 32 bits old Windows 7 or Windows 10.

reconbot, 2021-09-15:
Ouch, I hear you. How are you keeping node up to date since there don’t
ship a 32 but windows binary anymore?

SebCanet, 2021-09-15:
Dammit !!!!
I still see 32 bit version on their website (https://nodejs.org/en/download/current/), bu I'll try wth 64bit to verify this.
Thanks, I'll try.

zfangqijun, 2021-09-16:
[image]

I also encountered a similar problem.
Async loop get list,
plug and unplug usb multiple times,
get list will block

- serialport v9.04
- nodejs v12.20.0
- system macOS 10.15.7 (19H1217)
- electron v11.1.0
- hardware arduino uno

SebCanet, 2021-09-16:
> Dammit !!!!
> I still see 32 bit version on their website (https://nodejs.org/en/download/current/), but I'll try wth 64bit to verify this.
> Thanks, I'll try.

That's it ! All this updates have broke my 32bits compatibility... Now I have to decide: all version or 64 bits...

GazHank, 2021-09-29:
> [image]
> 
> I also encountered a similar problem. Async loop get list, plug and unplug usb multiple times, get list will block
> 
> * serialport v9.0.4
> * nodejs v12.20.0
> * system macOS 10.15.7 (19H1217)
> * electron v11.1.0
> * hardware arduino uno

hi @zfangqijun, based on your issue description, I think your issue is a little different, and might need to be logged as a new issue.  The list functionality on Windows, Mac and Linux are all quite different under the hood, so I suspect your issue could also be mac specific

closed 2022-04-28 by stale[bot]