serialport/node-serialport #2229: COM1 does not open for BUILT app; no issue in Windows VS Code Electron dev environment

Hugh-W50, 2021-03-15:
### Problem introduction
This issue was first alluded to here: https://stackoverflow.com/questions/66479424/what-diagnostics-does-windows-provide-to-assist-with-electron-app-failure-to-exe. In the absence of response, the following provides explicit definition of the problem hindering progress with app build/beta test process.

Having coded and debugged an Electron app using serialport for PTZ camera control, I need advice to identify __why the port__ used by the app  - once __built__ and __installed__ on this Windows 10 platform - __does not open/communicate__, causing the app (as coded) to report comms timeout.

It is confirmed that the app executes as expected in the VS Code/npm development environment, after working around this issue: https://stackoverflow.com/questions/66605483/why-does-node-serialport-fail-when-package-json-includes-postinstall-script.

Including use of serialport.parsers.Delimiter in the JS code (for main, not renderer), the only issue now preventing progress is lack of understanding to debug the failure of the port to open in the built app context.

### Summary of Problem
- What are you trying to do? Open COM1 under __built__ app serialport 9.0.7 control
- What happens? Electron app opens renderer window, button functions perform as expected, but camera communication fails with timeout, evidently due to port failure to open. There is no evidence of built app/OS exception reporting (app execution authorized in AVG firewall settings). Although I don't have a sniffer to hand, scripting Python to monitor the port is likely to confirm the failure; testing with another brand camera confirmed absence of serial comms.
- What should have happened? COM1 should open (as in VS Code dev environment), to enable camera comms and avoid timeout.

### Code to Reproduce the Issue
serialport-related code:
```
let cmdPacket = new Uint8Array(15) 
let recordTerminator = new Uint8Array(1)
recordTerminator[0] = 0xff
....
const port = new serialport('COM1',
                            { autoOpen: true,
                              baudRate: 9600,
                              stopBits: 1}//,
                            //((err) => {if (err){console.log('Error:', err)}else{console.log('COM1 opened!')}})
                            )
...
const Delimiter = serialport.parsers.Delimiter
const parser = port.pipe(new Delimiter({delimiter: recordTerminator}))
...
port.write(cmdPacket)
...
parser.on('data', (data) => {  ...}
```
package.json (same failure with alternate build script "electron-builder -w zip"):
```
    {
      "name": "alfatron",
      "version": "1.0.0",
      "description": "Proof of Concept",
      "main": "main.js",
      "scripts": {
         "start": "electron .",
         "watch": "nodemon --exec electron .",
         "reset": "git reset --hard",
         "build": "electron-builder -w nsis",
        "install": "electron-rebuild"
      },
     "repository": "https://bitbucket.org/hwylie_dev/alfatroncontroller1/src/main/",
     "keywords": [
     "Electron",
     "Alfatron",
     "prototype"
     ],
     "author": "H Wylie",
     "license": "CC0-1.0",
     "devDependencies": {
         "electron": "^11.2.0",
         "electron-builder": "^22.9.1",
         "electron-rebuild": "^2.3.5",
         "nodemon": "^2.0.0"
      },
      "dependencies": {
      "electron-window-state": "^5.0.3",
      "help": "^3.0.2",
      "serialport": "^9.0.7"
      }
    } 
```
Build log:
```
    > alfatron@1.0.0 build
    > electron-builder -w nsis

      • electron-builder  version=22.10.5 os=10.0.19041
      • electron-rebuild not required if you use electron-builder, please consider to remove excess dependency from
        devDependencies

    To ensure your native dependencies are always matched electron version, simply add script `"postinstall": "electron- 
    builder install-app-deps" to your `package.json`
      • rebuilding native dependencies  dependencies=@serialport/bindings@9.0.7 platform=win32 arch=x64
      • install prebuilt binary  name=@serialport/bindings version=9.0.7 platform=win32 arch=x64
      • packaging       platform=win32 arch=x64 electron=11.3.0 appOutDir=dist\win-unpacked
      • Unpacking electron zip  zipPath=undefined
      • default Electron icon is used  reason=application icon is not set
      • building        target=nsis file=dist\alfatron Setup 1.0.0.exe archs=x64 oneClick=true perMachine=false
      • building block map  blockMapFile=dist\alfatron Setup 1.0.0.exe.blockmap
```
### Versions, Operating System and Hardware

- SerialPort@9.0.7
- Node.js v12.18.3
- Windows Version 10.0.19041
- others: refer package.json above

- Hardware and chipset: Gigabyte GA-B75M-D3H Intel® B75 Express Chipset (COM1 motherboard header)
Processor	Intel(R) Core(TM) i5-3450 CPU @ 3.10GHz, 3501 Mhz, 4 Core(s), 4 Logical Processor(s)

#### Please advise steps recommended to identify and overcome this issue.
In particular, to confirm the fault lies with port.open() as opposed to port.write()/on.parser() in the __built__ context.

ecaii, 2021-03-16:
It's caused by electron-rebuild rebuilding the binaries: add `"npmRebuild": "false"` to your package.json, it should fix it.

Hugh-W50, 2021-03-16:
Inserting as recommended (at end of package.json) has no effect.
Deploying https://www.npmjs.com/package/electron-log to trace the built app (having inserted "log.info('About to open COM1')" and replaced console.log with log.info in the port constructor call, the log file is as follows:
```
    [2021-03-16 21:16:08.590] [info]  About to open COM1
    [2021-03-16 21:16:08.600] [info]  Error: Third argument must be a function
```
It's noted that the install script in package.json is essential, with or without "npmRebuild" insertion; without automatic rebuild invocation by 'npm i' results in dev (pre-build) 'npm run start' failure:
[npmRunStartLog.txt](https://github.com/serialport/node-serialport/files/6151760/npmRunStartLog.txt)

ecaii, 2021-03-17:
You should check this issue #2191

Hugh-W50, 2021-03-19:
Confirmed that forced rebuild of serialport@9.0.7 (@TonyStark106 advice of https://github.com/serialport/node-serialport/issues/2191) eliminated the error (npm v7.6.2).

closed 2021-03-19 by Hugh-W50