serialport/node-serialport #2234: electron-v87 support

linrong, 2021-04-07:
### Summary of Problem

Hi. I'm trying to build an electron app but during the install prebuilt binary process I'm getting messages like
```
 prebuild-install WARN install prebuilt binaries enforced with --force!
    prebuild-install WARN install prebuilt binaries may be out of date!
    prebuild-install info looking for local prebuild @ prebuilds\bindings-v9.0.7-electron-v87-win32-x64.tar.gz
    prebuild-install info looking for cached prebuild @ C:\Users\dgbuaa\AppData\Roaming\npm-cache\_prebuilds\e02868-bindings-v9.0.7-electron-v87-win32-x64.tar.gz
    prebuild-install http request GET https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@9.0.7/bindings-v9.0.7-electron-v87-win32-x64.tar.gz
    prebuild-install http 404 https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@9.0.7/bindings-v9.0.7-electron-v87-win32-x64.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=12.0.2 runtime=electron arch=x64 libc= platform=win32)
```
Quick check shows that there's just no release for Electron 12.0.2 as the highest is v85 but new Electron requires v87: https://github.com/serialport/node-serialport/releases

When will you add a support for Electron 12.0.2? It's a stable version now.
### Code to Reproduce the Issue

```js
const serialport = new SerialPort(path)

// Code
```

### Versions, Operating System and Hardware

- SerialPort@? 9.0.7
- Node.js v? 14.16
- Windows? Linux? Mac? windows
- Hardware and chipset? (Prolific/FTDI/Other)

samjonesigd, 2021-04-14:
Having this problem too

doubliez, 2021-04-28:
Same problem here. I tried Electron 11 but then encountered another issue (https://github.com/serialport/node-serialport/issues/2191), so for now I'm using Electron 10 which works fine.

hesa2020, 2021-05-01:
> Same problem here. I tried Electron 11 but then encountered another issue (#2191), so for now I'm using Electron 10 which works fine.

Which version of node are you using ?

samjonesigd, 2021-05-04:
For anyone reading I eventually got this working by using electron 10 and serialport 9.0.3 & rebuilding with electron-rebuild

devrim-oguz, 2021-05-27:
Install electron-rebuild from npm and build the node modules using it.

doubliez, 2021-06-01:
I'm building Windows installers using the Docker image `electronuserland/builder:wine` and it seems electron-rebuild doesn't help in this case, it always tries to download pre-built binaries...

Any plans on adding pre-built binaries for electron-v87?

GazHank, 2021-06-06:
Hi @doubliez

Since your container image uses electron builder you might want to try:

Add "buildDependenciesFromSource": true,"npmRebuild": false, to your project.json's build configuration

This should help until electron 12 is prebuilt

Cheers

Gaz

closed 2021-06-18 by reconbot

reconbot, 2021-06-19:
Please try `serialport@9.2.0`

PR #2274 (2021-06-18): fix: electron 13 prebuilds
Simply updates node-abi

This will make the next release produce prebuilds for electron 13, electron 12 and node 16

fixes #2234 fixes #2268