serialport/node-serialport #2238: PR: Unify API between WebSerial and this

frank-dspeed, 2021-04-20:
## 💥 Proposal

### What feature you'd like to see
(A clear and concise description of what the proposal is.)
I will implement functions that exist on the WebSerial API on this Project as also Port the most API's of this project to WebSerial.

## Motivation
(Please outline the motivation for the proposal. It's interesting knowing what people are working on and also could help community members make suggestions for work-arounds until the feature is built)

I want to enable Easyer Code ReUse between the both API's so that serialport and webSerialApi can be used with the same code.

## Pitch

(Please explain why this feature should be implemented and how it would be used.)

Will Allow people to use node-serialport as normal with NodeJS while allowing to switch to WebSerialAPI when running in Electron or NW.JS or the Browser.

## Implementation State
As always i will implement that my self 👯‍♂️ 

### Related Issues
- SerialPort.list() doesn't work anymore? #2228
-

frank-dspeed, 2021-04-20:
```
navigator.serial.getPorts().then((ports)
serialport.list() alias getPorts

## Adding API on SerialPort draft-1
.getPorts
.requestPort({ filters: [{ usbVendorId }]})

reconbot, 2021-04-20:
Making a binding that uses web serial is a great idea.

frank-dspeed, 2021-04-24:
## Conclusion
Implementing a WebSerial API brings no reuseable code because of the Permission System of the Browser and so the alternate methods of retriving ports.

There exists a WebStream Ponyfill that can get used to reuse Logic between the implementations.

At All i Come to the conclusion that nothing can be added here that would bring any benefits anyway i want to add similar behavior to some of the functions in here i think needs more evaluations of code samples

reconbot, 2021-07-05:
closing this over #2271

closed 2021-07-05 by reconbot